adafruit/circuitpython #10266: Feather RP2040 with USB Host doesn't recognize other RP2040 boards

rianadon, 2025-04-17:
### CircuitPython version and board name

```python
Adafruit CircuitPython 9.2.7 on 2025-04-01; Adafruit Feather RP2040 USB Host with rp2040
```

### Code/REPL

```python
import usb
list(usb.core.find(find_all=True))
```

### Behavior

Returns an empty array

### Description

I'm trying to connect another RP2040 board with CircuitPython to the Feather RP2040 USB Host via the USB connection. However, as I've put in the Code/Behavior sections, the Feather is not finding this board.

The USB device I'm plugging in is specifically a WaveShare RP2040-Zero with the CircuitPython 9.2.7.

Other things I've tried are:
- Plugging in other USB devices to the Feather. These work.
- Putting the RP2040-Zero with BOOT held as I plug it in. CircuitPython doesn't recognize the bootloader either.
- Running Arduino on the Feather instead and using the Adafruit TinyUSB examples/DualRole/simple/device_info sketch. This works if I increase `CFG_TUH_ENUMERATION_BUFSIZE`.
- Running with TinyUSB in debug mode. The process hangs near the beginning of device enumeration. The point at which it hangs seems to vary slightly, but it usually gets stuck after the "Get Descriptor" stage.
- Running with another RP2040 repurposed as a logic analyzer. After enumeration fails, there seems to be a forever repeating STATUS packet with a NAK.

rianadon, 2025-04-17:
I flashed https://github.com/ataradov/usb-sniffer-lite onto a spare Pico and got these results which aren't too revealing as to what the problem could be:

CircuitPython:
```
1381732 : --- RESET ---
   ... : Folded 450 frames
  1000 : SOF #504
    16 : SETUP: 0x00/0
     3 : DATA0 (8): 80 06 00 01 00 00 08 00 
     9 : ACK
   ... : Folded 7 frames
  1000 : SOF #512
    16 : IN: 0x00/0
     2 : DATA1 (8): 12 01 00 02 00 00 00 40 
    10 : ACK
   972 : SOF #513
    16 : OUT: 0x00/0
     3 : DATA1: ZLP
     3 : NAK
   978 : SOF #514
    16 : OUT: 0x00/0
     4 : DATA1: ZLP
     3 : NAK
```
The SOF/OUT/DATA1/NAK repeat indefinitely.

Arduino:
```
1533127 : --- RESET ---
   ... : Folded 458 frames
  1000 : SOF #1329
    17 : SETUP: 0x00/0
    20 : DATA0 (8): 80 06 00 01 00 00 08 00 
    29 : ACK
   ... : Folded 1 frame
  1000 : SOF #1331
    18 : IN: 0x00/0
    20 : DATA1 (8): 12 01 00 02 00 00 00 40 
    30 : ACK
   ... : Folded 9 frames
  1001 : SOF #1341
    18 : OUT: 0x00/0
    21 : DATA1: ZLP
    24 : ACK
   ... : Folded 7 frames
  1000 : SOF #1349
    18 : SETUP: 0x00/0
    21 : DATA0 (8): 00 05 01 00 00 00 00 00 
    30 : ACK
```
The device ACKS and the Arduino host proceeds to change the USB address.

tannewt, 2025-04-21:
Try absolute newest CircuitPython. I believe it has newer TinyUSB in it and we've been fixing things like this.

rianadon, 2025-04-22:
Thanks! I should've tried the alpha release before asking. I also discovered my local CircuitPython git was a few weeks behind.

Both the 10.0.0-alpha.2 and main branch work when plugging in my RP2040 devices, even with CircuitPython, save for one...
On one of the CircuitPython devices I enabled the USB CDC data stream (`usb_cdc.enable(console=True, data=True)`). The configuration descriptor for this device is now 284 bytes, which exceeds the `CFG_TUH_ENUMERATION_BUFSIZE=256` set in `supervisor/shared/usb/tusb_config.h`. Would you consider increasing the size of this buffer?

tannewt, 2025-04-22:
Sure! Feel free to make a PR. Thanks for the testing!

closed 2025-04-22 by rianadon