adafruit/circuitpython #10268: ESP32S2 fails to read files from mounted SD-card

bablokb, 2025-04-17:
### CircuitPython version and board name

```python
Adafruit CircuitPython 9.2.7 on 2025-04-01; S2Mini with ESP32S2-S2FN4R2
```

### Code/REPL

```python
import supervisor
import time
import board
import audiomp3
import audiobusio
import busio

import sdcardio
import storage
import os

# --- configuration   ---

# SD-card (SPI)
PIN_MISO = board.IO10
PIN_MOSI = board.IO6
PIN_CLK  = board.IO8
PIN_CS   = board.IO4
PINS_I2S = [board.IO11,board.IO12,board.IO9] # BLCK, WSEL, DATA

# --- mount SD-card   ---

def mount_sd():
  """ mount SD-card """

  spi = None
  try:
    spi    = busio.SPI(PIN_CLK,PIN_MOSI,PIN_MISO)
    sdcard = sdcardio.SDCard(spi,PIN_CS,1_000_000)
    vfs    = storage.VfsFat(sdcard)
    storage.mount(vfs, "/mnt")
  except Exception as ex:
    if spi:
      spi.deinit()
    raise
  return spi

# --- play music   ------

def play_music(files):
  """ create music task """
  audio = audiobusio.I2SOut(*PINS_I2S)

  print("start playing music")
  for f in files:
    try:
      fs = open(f,"rb")
      mp3 = audiomp3.MP3Decoder(fs)
      print(f"starting {f}")
      audio.play(mp3)
      while audio.playing:
        time.sleep(0.5)
      print(f"finished {f}")
    except Exception as ex:
      print(f"could not play {f}: {ex}")
    fs.close()
    mp3.deinit()
    time.sleep(0.5)
  print("done playing")
  
# --- main program   ----

while not supervisor.runtime.serial_connected:
  time.sleep(1)

spi = mount_sd()
files = os.listdir("/mnt")
print(f"os.listdir('/mnt'): {files=}")
files = ["/mnt/"+f for f in files if f[-3:] == "mp3"]
files.sort()
print(f"sorted: {files=}")

play_music(files)
```

### Behavior

```
os.listdir('/mnt'): files=['music2.mp3', 'music1.mp3', 'music3.mp3', 'music4.mp3']
sorted: files=['/mnt/music1.mp3', '/mnt/music2.mp3', '/mnt/music3.mp3', '/mnt/music4.mp3']
start playing music
starting /mnt/music1.mp3
could not play /mnt/music1.mp3: [Errno 5] Eingabe-/Ausgabefehler
could not play /mnt/music2.mp3: [Errno 5] Eingabe-/Ausgabefehler: /music2.mp3
could not play /mnt/music3.mp3: [Errno 5] Eingabe-/Ausgabefehler: /music3.mp3
could not play /mnt/music4.mp3: [Errno 5] Eingabe-/Ausgabefehler: /music4.mp3
done playing
 
Programm wird ausgeführt.
espidf.IDFError: 

### Description

The program mounts a SD-card, queries all the mp3 files on the card (using `os.listdir()`) and then plays one file after the other. Fairly straightforward and perfectly working on a Pico (9.2.6) with the same SD-card and SD-breakout.

Things to note:

  - the device (Wemos Lolin S2-Mini) has no problems playing the files when loading them from internal flash
  - I also tried 9.1.4, same problem
  - The error message is a bit confusing: it seems that CP strips the mount point and tries to read the (non-existent) files from the root fs. But the error code is 5 (EIO) and not 2 (ENOENT)
  - Stripping of the mount-point is systematic, I tried `/sd`, `/mnt`, `/foo/bar`.

### Additional information

bablokb, 2025-04-22:
This seems to be a problem with DMA-buffers being overwritten from I2S side thus corrupting memory for SPI. I managed to fix this with this change:

```
diff --git a/ports/espressif/common-hal/audiobusio/__init__.c b/ports/espressif/common-hal/audiobusio/__init__.c
index 226e371c5b..d07a0b521b 100644
--- a/ports/espressif/common-hal/audiobusio/__init__.c
+++ b/ports/espressif/common-hal/audiobusio/__init__.c
@@ -184,7 +184,7 @@ void port_i2s_play(i2s_t *self, mp_obj_t sample, bool loop) {
         self->next_buffer_size = sizeof(starting_frame);
         i2s_fill_buffer(self);
         i2s_channel_preload_data(self->handle, &starting_frame, sizeof(uint32_t), &bytes_loaded);
-        preloaded += 1;
+        preloaded += bytes_loaded;
     }
 
     // enable the channel
```

Maybe someone who knows better wants to check this. I am happy to add this as a PR if you think this makes sense.

tannewt, 2025-04-22:
Looks more correct to me! Please make a PR. Thanks!

bablokb, 2025-04-24:
Fixed with #10284

closed 2025-04-24 by bablokb