adafruit/circuitpython #10295: Audio stopping during playback when using audiofreeverb.Freeverb

relic-se, 2025-04-25:
### CircuitPython version and board name

```python
Adafruit CircuitPython 10.0.0-alpha.2-38-g8a28033587 on 2025-04-25; Pimoroni Pico Plus 2 with rp2350b
```

### Code/REPL

```python
import time
import board
import busio
import audiobusio, audiocore, audiomixer, synthio
import audiofreeverb, audiofilters

CHANNELS = 2
SRATE = 48000

i2s_bclk, i2s_lclk, i2s_data = board.GP2, board.GP3, board.GP4
audio = audiobusio.I2SOut(bit_clock=i2s_bclk, word_select=i2s_lclk, data=i2s_data)

mixer = audiomixer.Mixer(voice_count=1, channel_count=CHANNELS, sample_rate=SRATE, buffer_size=2048)
audio.play(mixer)

synth = synthio.Synthesizer(channel_count=CHANNELS, sample_rate=SRATE)

effect = audiofreeverb.Freeverb(channel_count=CHANNELS, sample_rate=SRATE)
effect.play(synth)

lpf = audiofilters.Filter(channel_count=CHANNELS, sample_rate=SRATE, mix=1.0)
lpf.filter = synthio.Biquad(synthio.FilterMode.LOW_PASS, frequency=1000, Q=1.0)

lpf.play(effect)
mixer.voice[0].play(lpf)

for i in range(24):
    print(i + 1)
    synth.press(synthio.Note(synthio.midi_to_hz(36 + i * 3), amplitude=0.5))
    time.sleep(0.5)
```

### Behavior

Program continues running, but audio ceases altogether when the 10th note is pressed. If the filter and its module is removed, the audio cuts off at the 18th note.

### Description

If I add in a `print(gc.mem_free())` statement to the for loop, I get 8258240 bytes available at the note that cuts off (the board has 8MB of PSRAM).

### Additional information

I initially thought the error was due to internal memory being filled up, but I'm unable to recreate the issue in versions of CircuitPython prior to the latest dev build without `audiofreeverb.Freeverb`. I think the error may lie somewhere within that effect.

gamblor21, 2025-04-25:
I have to look more but if you aren't releasing the notes `synthio` can only play so many at once. Does this occur with any other effects?

relic-se, 2025-04-26:
The limit on RP2350 is 24 notes. Under normal circumstances, the notes are either ignored or the oldest notes are dropped. In this situation, audio stops altogether.

relic-se, 2025-04-26:
I've tested it with `audiodelays.Echo`, but was unable to recreate the issue.

gamblor21, 2025-04-27:
I tried this on the latest nightly build (Adafruit CircuitPython 10.0.0-alpha.2-40-gfa9b9746f9 on 2025-04-26; Raspberry Pi Pico 2 with rp2350a) and it worked normal for me. All notes played.

As I typed this I tried it a couple more times and had inconsistent results. This may be related to the issue being worked on when audio DMAs are not working. I cannot think of any part of Freeverb that would cause audio to stop in this manner as it does not base any functionality on Notes just the audio information it takes in.

tannewt, 2025-04-28:
I'm working to fix an issue where the audio buffers stop ping ponging to each other. This can happen if the second buffer completes while the first is being filled.

I suspect that the time it takes to compute the audio buffer may influence whether this happens. The longer it takes to fill the buffer, the more likely it is to happen.

tannewt, 2025-05-01:
Please retest with https://github.com/adafruit/circuitpython/pull/10299 included. It tweaks how the buffers ping pong back and forth.

relic-se, 2025-05-02:
@tannewt Yep, that fixes the problem I was running into. The provided demo works without the error now. 💯

closed 2025-05-02 by relic-se