adafruit/circuitpython #10296: Feather ESP32-C6 Fails with code.circuitpython.org USB Connection

eightycc, 2025-04-26:
### CircuitPython version and board name

```python
Adafruit CircuitPython 10.0.0-alpha.2 on 2025-04-04; Adafruit Feather ESP32-C6 4MB Flash No PSRAM with ESP32C6
```

### Code/REPL

```python
n/a
```

### Behavior

USB connection using `code.circuitpython.org` succeeds, but directory displays garbage. Subsequently disconnecting from `code.circuitpython.org`, connecting with `minicom`, then hitting `ctrl-D` results in:
```
soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```

### Description

Testing with CP 9.2.7 succeeds. `code.circuitpython.org` works as expected. No crash occurs.

eightycc, 2025-04-26:
Additional notes:

- Flash completely erased between tests
- Similar behavior with Thonny

eightycc, 2025-04-26:
Bisecting 62 commits from tags `9.2.7` to `10.0.0-alpha.2`.

eightycc, 2025-04-26:
Regression introduced in this commit: https://github.com/adafruit/circuitpython/pull/10208.

eightycc, 2025-04-26:
Regression introduced in this PR: https://github.com/adafruit/circuitpython/pull/10208

eightycc, 2025-04-27:
`code.circuitpython.org` sends a small Python program to display the root directory:
```python
import os
import time
contents = os.listdir("/")
for item in contents:
    result = os.stat("/" + item)
    print(item, result[0], result[6], result[9])
```

This triggers one of two failures, one is trapped by a hardware breakpoint on `_panic_handler`, the other is trapped by a hardware breakpoint on `mp_obj_new_exception_msg`.
```
(gdb) hb mp_obj_new_exception_msg
Hardware assisted breakpoint 2 at 0x4200fcd4: file ../../py/objexcept.c, line 491.
(gdb) hb _panic_handler
Hardware assisted breakpoint 3 at 0x40800102: file /Users/rabeles/Development/cp-development/issue-10296/cp-repo-10296/ports/espressif/esp-idf/components/riscv/vectors.S, line 141.
```
The `mp_obj_new_exception_msg` case:
```
Thread 4 "main" hit Breakpoint 2, mp_obj_new_exception_msg (exc_type=0x4215d8f0 <mp_type_NotImplementedError>, msg=0x421603e4 <translation464>)
    at ../../py/objexcept.c:491
491	    return mp_obj_new_exception_msg_varg(exc_type, msg);
(gdb) bt
#0  mp_obj_new_exception_msg (exc_type=0x4215d8f0 <mp_type_NotImplementedError>, msg=0x421603e4 <translation464>) at ../../py/objexcept.c:491
#1  0x4201a5d8 in mp_execute_bytecode (code_state=code_state@entry=0x408403fc, inject_exc=inject_exc@entry=0x0)
    at build-adafruit_feather_esp32c6_4mbflash_nopsram/genhdr/qstrdefs.generated.h:1977
#2  0x420107c2 in fun_bc_call (self_in=0x40840df0, n_args=<optimized out>, n_kw=0, args=0x0) at ../../py/objfun.c:271
#3  0x42057c34 in parse_compile_execute (source=source@entry=0x40836f30, input_kind=input_kind@entry=MP_PARSE_FILE_INPUT, 
    exec_flags=exec_flags@entry=17, result=result@entry=0x0) at ../../shared/runtime/pyexec.c:161
#4  0x42057e54 in pyexec_raw_repl () at ../../shared/runtime/pyexec.c:628
#5  0x4201faf6 in run_repl (safe_mode=SAFE_MODE_NONE) at ../../main.c:944
#6  main () at ../../main.c:1103
(gdb) frame 1
#1  0x4201a5d8 in mp_execute_bytecode (code_state=code_state@entry=0x408403fc, inject_exc=inject_exc@entry=0x0)
    at build-adafruit_feather_esp32c6_4mbflash_nopsram/genhdr/qstrdefs.generated.h:1977
1977	TRANSLATION("opcode", 464)
(gdb) info locals
obj = <optimized out>
ip = 0x40840f4e "\204@lib"
sp = 0x40840410
qstr_table = 0x40840e50
obj_shared = <optimized out>
nlr = {prev = 0x40836db0, ret_val = 0x1, jmpbuf = {4648845097327239222, 4648337483707452912, 0, 12884901898, 4, 0, 4648677850193330176, 
    4755954091266444018, 4648282800183798976, 2148566003728, -4294967281, 4756283206020478656, 4648282800183799008, 4612545605674430480, 
    4648856039796506639, 1082393596, 1082393616, 4648845250838659072, 0, 4294967298, 0, 4648845182119182346, 4648856039796506624, 1082393652, 
    7, 271665213440, 4648339234971713539, 4756091117903132018, 1082355040, 171798691840, 10, 4648856040878900220, 1082393596, 0, 
    4648845372180018768, 4294967298, 1082275408, 0}, pystack = 0x4084043c}
entry_table = {0x4201a5c4 <mp_execute_bytecode+1510> <repeats 16 times>, 0x4201a5e6 <mp_execute_bytecode+1544>, 
  0x4201a386 <mp_execute_bytecode+936>, 0x4201a528 <mp_execute_bytecode+1354>, 0x4201b18c <mp_execute_bytecode+4526>, 
  0x4201a44a <mp_execute_bytecode+1132>, 0x4201b158 <mp_execute_bytecode+4474>, 0x4201b088 <mp_execute_bytecode+4266>, 
  0x4201b008 <mp_execute_bytecode+4138>, 0x4201b03a <mp_execute_bytecode+4188>, 0x4201af46 <mp_execute_bytecode+3944>, 
  0x4201af14 <mp_execute_bytecode+3894>, 0x4201a556 <mp_execute_bytecode+1400>, 0x4201a72e <mp_execute_bytecode+1872>, 
  0x4201a5c4 <mp_execute_bytecode+1510>, 0x4201a5c4 <mp_execute_bytecode+1510>, 0x4201a5c4 <mp_execute_bytecode+1510>, 
  0x4201a920 <mp_execute_bytecode+2370>, 0x4201a8bc <mp_execute_bytecode+2270>, 0x4201a65a <mp_execute_bytecode+1660>, 
  0x4201a4e0 <mp_execute_bytecode+1282>, 0x4201a496 <mp_execute_bytecode+1208>, 0x4201b1e0 <mp_execute_bytecode+4610>, 
  0x4201b0f0 <mp_execute_bytecode+4370>, 0x4201b0ba <mp_execute_bytecode+4316>, 0x4201afaa <mp_execute_bytecode+4044>, 
  0x4201af74 <mp_execute_bytecode+3990>, 0x4201ab88 <mp_execute_bytecode+2986>, 0x4201ab5a <mp_execute_bytecode+2940>, 
  0x4201ab34 <mp_execute_bytecode+2902>, 0x4201aaf6 <mp_execute_bytecode+2840>, 0x4201aab6 <mp_execute_bytecode+2776>, 
  0x4201aa30 <mp_execute_bytecode+2642>, 0x4201aa6c <mp_execute_bytecode+2702>, 0x4201a9da <mp_execute_bytecode+2556>, 
  0x4201a9ac <mp_execute_bytecode+2510>, 0x4201a95a <mp_execute_bytecode+2428>, 0x4201a87c <mp_execute_bytecode+2206>, 
  0x4201a83c <mp_execute_bytecode+2142>, 0x4201a246 <mp_execute_bytecode+616>, 0x4201a7fc <mp_execute_bytecode+2078>, 
  0x4201a5c4 <mp_execute_bytecode+1510>, 0x4201a5c4 <mp_execute_bytecode+1510>, 0x4201a5c4 <mp_execute_bytecode+1510>, 
  0x4201a5c4 <mp_execute_bytecode+1510>, 0x4201a5c4 <mp_execute_bytecode+1510>, 0x4201a5c4 <mp_execute_bytecode+1510>, 
  0x4201a5c4 <mp_execute_bytecode+1510>, 0x4201a5c4 <mp_execute_bytecode+1510>, 0x4201ac7a <mp_execute_bytecode+3228>, 
  0x4201a5c4 <mp_execute_bytecode+1510>, 0x4201ae60 <mp_execute_bytecode+3714>, 0x4201ae38 <mp_execute_bytecode+3674>, 
  0x4201ae10 <mp_execute_bytecode+3634>, 0x4201ad76 <mp_execute_bytecode+3480>, 0x4201ad54 <mp_execute_bytecode+3446>, 
  0x4201ad98 <mp_execute_bytecode+3514>, 0x4201a2be <mp_execute_bytecode+736>, 0x4201a2be <mp_execute_bytecode+736>, 
  0x4201ac2a <mp_execute_bytecode+3148>, 0x4201abb6 <mp_execute_bytecode+3032>, 0x4201a5c4 <mp_execute_bytecode+1510>, 
[... cut ...]
#13 0x42057e54 in pyexec_raw_repl () at ../../shared/runtime/pyexec.c:628
628	        int ret = parse_compile_execute(&line, MP_PARSE_FILE_INPUT, EXEC_FLAG_PRINT_EOF | EXEC_FLAG_SOURCE_IS_VSTR, NULL);
(gdb) info locals
ret = 1082396774
line = {alloc = 176, len = 155, 
  buf = 0x40840d40 "\nimport os\nimport time\ncontents = os.listdir(\"/\")\nfor item in contents:\n    result = os.stat(\"/\" + item)\n    print(item, result[0], result[6], result[9])\n\n", fixed_buf = false}
raw_repl_reset = <optimized out>
(gdb) 

```

tannewt, 2025-04-28:
Please retest this with this fix: https://github.com/adafruit/circuitpython/pull/10291

dhalbert, 2025-04-28:
This is subsumed by #10298.

closed 2025-04-28 by dhalbert