adafruit/circuitpython #10298: error in paste mode

dhalbert, 2025-04-28:
Reproducible on ESP32-C3 and ESP32-C6:

A code error during paste mode or raw mode causes a hard crash when exiting the mode with ctrl-D.

Simplest example:
```

eightycc, 2025-05-03:
The workaround described above stopped working with the introduction of gcc 14.2.0 by ESP-IDF 5.4.1. The reason this happened was a change in the ordering of generated RISC-V instructions in function `parse_compile_execute` when compiled with gcc 14.2.0:

```
0x42057e62 <parse_compile_execute+100>:	lw	a4,28(sp)
   0x42057e64 <parse_compile_execute+102>:	li	a5,1
   0x42057e66 <parse_compile_execute+104>:	lw	s3,0(s1)
   0x42057e6a <parse_compile_execute+108>:	beq	a4,a5,0x42057f90 <parse_compile_execute+402>
   0x42057e6e <parse_compile_execute+112>:	lw	a1,28(sp)
   0x42057e70 <parse_compile_execute+114>:	mv	a0,s1
   0x42057e72 <parse_compile_execute+116>:	jal	0x42004baa <mp_parse>
   0x42057e76 <parse_compile_execute+120>:	lw	a5,16(sp)
   0x42057e78 <parse_compile_execute+122>:	sw	a0,56(sp)
   0x42057e7a <parse_compile_execute+124>:	sw	a1,60(sp)
   0x42057e7c <parse_compile_execute+126>:	srli	a2,a5,0x2
   0x42057e80 <parse_compile_execute+130>:	andi	a2,a2,1
   0x42057e82 <parse_compile_execute+132>:	mv	a1,s3
   0x42057e84 <parse_compile_execute+134>:	addi	a0,sp,56
   0x42057e86 <parse_compile_execute+136>:	jal	0x420082da <mp_compile>
   0x42057e8a <parse_compile_execute+140>:	lw	a4,28(sp)
   0x42057e8c <parse_compile_execute+142>:	li	a5,1
   *** Pointer to compiled function returned by mp_compile copied to s1 ***
   0x42057e8e <parse_compile_execute+144>:	mv	s1,a0
   0x42057e90 <parse_compile_execute+146>:	bne	a4,a5,0x42057e98 <parse_compile_execute+154>
   0x42057e94 <parse_compile_execute+150>:	jal	0x4201fb8e <gc_collect>
   *** Store into module_fun on stack frame after call to gc_collect ***
   0x42057e98 <parse_compile_execute+154>:	sw	s1,24(sp)
   0x42057e9a <parse_compile_execute+156>:	li	a0,3
   0x42057e9c <parse_compile_execute+158>:	jal	0x42057d64 <mp_hal_set_interrupt_char>
   0x42057ea0 <parse_compile_execute+162>:	beqz	s2,0x42057fa2 <parse_compile_execute+420>
   0x42057ea4 <parse_compile_execute+166>:	lw	a5,12(sp)
   0x42057ea6 <parse_compile_execute+168>:	lw	a3,12(a5)
   0x42057ea8 <parse_compile_execute+170>:	lw	a2,4(a5)
   0x42057eaa <parse_compile_execute+172>:	lw	a1,0(a5)
   0x42057eac <parse_compile_execute+174>:	lw	a0,8(a5)
   0x42057eae <parse_compile_execute+176>:	jal	0x4200aa94 <mp_call_function_n_kw>

```

The call to `mp_compile` returns a pointer to the compiled function object in `a0`. `a0` is copied to `s1` and then `gc_collect` is called. Subsequent to calling `gc_collect`, `s1` is stored into the stack frame. Because the pointer to the compiled function is not yet in the stack frame when `gc_collect` is called, `gc_collect` frees it, resulting in the memory occupied by the executing function's compiled code getting re-allocated and overlayed while it is executing.

Making `module_fun` `volatile` prevents gcc's RISC-V codegen from delaying its store into the stack frame, durably correcting the issue.

eightycc, 2025-05-06:
Fixed by #10311.

closed 2025-05-06 by eightycc