adafruit/circuitpython #10300: Oxocard Galaxy: Display initialization problem

HilmarSt, 2025-04-30:
### CircuitPython version and board name

```python
Adafruit CircuitPython 9.2.7 on 2025-04-01; Oxocard Galaxy with ESP32
Board ID:oxocard_galaxy
```

### Code/REPL

```python
CircuitPython: board.c
display_init(void)
```

### Behavior

After Power-up, the upper third of the Display shows random pixels (not initialized).

### Description

When flashing the Oxocard Galaxy with the binary for "Oxocard Connect", the display is working as expected.
The difference in _display_init()_ is only one line in _common_hal_busdisplay_busdisplay_construct_ :
Galaxy (fail): `80,             // row start`
Connect (correct): `0,             // row start`

@supcik:
Maybe you can do this tiny fix in a minute...

supcik, 2025-05-01:
Sure, I will test it on a recent oxocard and fix it

supcik, 2025-05-01:
I was able to replicate the problem :

[image].

supcik, 2025-05-01:
I opened the PR #10302 to fix this issue.

tannewt, 2025-05-01:
Fixed by #10302. Thank you both!

closed 2025-05-01 by tannewt