adafruit/circuitpython #10307: I2C peripheral in use after soft reset

jedgarpark, 2025-05-02:
### CircuitPython version and board name

```python
Adafruit CircuitPython 10.0.0-alpha.3 on 2025-05-02; Adafruit Fruit Jam with rp2350b
```

### Code/REPL

```python
Larsio Paint Music https://github.com/adafruit/Adafruit_Learning_System_Guides/pull/3013
```

### Behavior

after soft reset  (ctrl-D, saving file, etc.) `Error initializing TLV320 DAC: I2C peripheral in use` Hard reset will fix it.

jedgarpark, 2025-05-02:
@tannewt I put the issue we discussed here

todbot, 2025-05-02:
I am also seeing this on `Adafruit CircuitPython 10.0.0-alpha.3 on 2025-05-02; Raspberry Pi Pico with rp2040` with the sample code below.   On reset, the audio plays. Then if you Ctrl-C and Ctrl-D to restart, the error following error appears: 
```Traceback (most recent call last):
  File "code.py", line 14, in <module>
ValueError: I2C peripheral in use
```

This does not happen on `Adafruit CircuitPython 10.0.0-alpha.2 on 2025-05-02; Raspberry Pi Pico with rp2040`

Sample code: 
```py
import time, math, time
import ulab.numpy as np
import board, busio, audiobusio, audiocore, audiomixer
import adafruit_tlv320 

SAMPLE_RATE = 44100
CHANNEL_COUNT = 1
BUFFER_SIZE = 2048

i2c_sda_pin, i2c_scl_pin = board.GP16, board.GP17
i2s_bck_pin, i2s_lck_pin, i2s_dat_pin = board.GP20, board.GP21, board.GP22

i2c = busio.I2C(scl=i2c_scl_pin, sda=i2c_sda_pin)
audio = audiobusio.I2SOut(bit_clock=i2s_bck_pin, word_select=i2s_lck_pin, data=i2s_dat_pin)
dac = adafruit_tlv320.TLV320DAC3100(i2c)
dac.configure_clocks(sample_rate=SAMPLE_RATE, bit_depth=16)
dac.headphone_output = True
dac.headphone_volume = -35  # dB, does not seem to work? 

mixer = audiomixer.Mixer(sample_rate=SAMPLE_RATE, channel_count=CHANNEL_COUNT, buffer_size=BUFFER_SIZE)
audio.play(mixer)

# generate a sine wave
tone_volume = 0.5
frequency = 440
length = SAMPLE_RATE // frequency
sine_wave = np.array(np.sin(np.linspace(0, 2*np.pi, length, endpoint=False))
                     * tone_volume * 32767, dtype=np.int16)
sine_wave_sample = audiocore.RawSample(sine_wave, sample_rate=SAMPLE_RATE)

mixer.voice[0].play(sine_wave_sample, loop=True)

while True:
    for db in range(-60, 0, 5):
        print("setting headphone db:", db)
        dac.headphone_volume = db
        time.sleep(0.5)
```

relic-se, 2025-05-03:
I'm having this error on `Adafruit CircuitPython 10.0.0-alpha.3 on 2025-05-02; Pimoroni Pico Plus 2 with rp2350b` with the following code:
```
import board, busio
i2c = busio.I2C(scl=board.GP19, sda=board.GP18)
```
Which results in:
```
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
ValueError: I2C peripheral in use
```
The board-specific I2C is GP4/GP5 which is I2C0, whereas GP18/GP19 is I2C1. `i2c = board.I2C()` works without error.

relic-se, 2025-05-03:
Brute force testing revealed that `adafruit-circuitpython-...-20250426-main-PR10264-....uf2` is the first build to trigger this error. It must stem from the changes in #10264, and is related to garbage collection? Hopefully that'll get you something to start with, @tannewt.

tannewt, 2025-05-03:
Please try this branch: https://github.com/tannewt/circuitpython/tree/fix_finalisers

I think I broke finalisers in that change. I only have time to compile it now.

todbot, 2025-05-03:
Yep, seems to have fixed it. My above test code on RP2040 works a second time after Ctrl-C, Ctrl-D, as well as a version of @relic-se's much simpler example of just creating a `busio.I2C()`.

jedgarpark, 2025-05-03:
yes, @tannewt this is fixed for me now too. Fruit Jam sucessfully re-initializing DAC on I2C after soft resets (both from file save and ctrl-D).

closed 2025-05-04 by gamblor21

relic-se, 2025-05-05:
Sorry for the late response, but the alpha.4 update fixes the bug for me too. Thanks @tannewt !