adafruit/circuitpython #10310: board.os_id

b-blake, 2025-05-05:
Hi,

Can you add a function under board where the user can get the description of the CP OS in use?

board.os_id looks a lot like board.board_id to keep format.  Both are derrived from the OS.

Bruce

Neradoc, 2025-05-05:
Hi, that's not information related to the board, it is found in the os and sys modules.
For example:
```py
>>> import os
>>> os.uname()
(sysname='ESP32S3', nodename='ESP32S3', release='10.0.0', version='10.0.0-alpha.2-11-gb9f631e219 on 2025-04-14', machine='ESP32-S3-Box-2.5 with ESP32S3')
>>> import sys
>>> sys.implementation
(name='circuitpython', version=(10, 0, 0, ''), _machine='ESP32-S3-Box-2.5 with ESP32S3', _mpy=774)
```

b-blake, 2025-05-05:
@Neradoc,

OH,  I didn't know that.  Thank you.

Bruce

closed 2025-05-05 by tannewt