adafruit/circuitpython #10326: Raspberry Pi Pico CircuitPython COM port data support

bernard01, 2025-05-09:
CircuitPython has support for a single COM serial port.

On the Pico, the only way I see to get it is to sacrifice the console:

boot.py
import usb_cdc
usb_cdc.enable(console=False, data=True)    # Disable console, enable data

That's a very awkward solution because programming MicroPython without a console is too hard.

Somewhere the documentation says that there are difficulties in this area with some devices, without giving specifics.

So I don't even know which device to buy to get the required support.

bablokb, 2025-05-09:
You could create an UART and attach an UART-to-USB cable. This will allow to send data and still use the REPL on the USB- console.

bernard01, 2025-05-09:
> You could create an UART and attach an UART-to-USB cable. This will allow to send data and still use the REPL on the USB- console.

Thanks. That's a fairly good idea. I think I could even use a native serial port. But I want the USB serial data port single cable solution.

Neradoc, 2025-05-09:
But I don't understand what your issue is, you intentionally disable the console to reduce the number of com ports ?
But you also want the console at the same time ?

This is supported on RP2040:
```py
import usb_cdc
usb_cdc.enable(console=True, data=True)
```

The ESP32 S2 and S3 for example (and some other ports) don't have enough USB endpoints to support the drive, HID and 2 serials at the same time, that's why there is that warning.

bernard01, 2025-05-09:
> I don't understand what your issue is, you intentionally disable the console to reduce the number of com ports ? But you also want the console at the same time ?
> 
> This is supported on RP2040:
> 
> import usb_cdc
> usb_cdc.enable(console=True, data=True)
> 
Not intentionally. Not my choice. While it MAY be supported as you write on some RP2040 devices which I don't know what they are, it is not supported on the Pico. I cannot see a single piece of evidence on the web hat shows how it works on it. Instead, I see some articles that are looking for solutions. One of them is 
[Possible to send data from a Pi Pico via USB while running CircuitPython?](https://raspberrypi.stackexchange.com/questions/135422/possible-to-send-data-from-a-pi-pico-via-usb-while-running-circuitpython)

Neradoc, 2025-05-09:
> it is not supported on the Pico

Yes it is, on EVERY raspberry pi board, since this is unrelated to the board, only to the chip's USB support.

It's true that the use of usb_cdc kind of lacks a dedicated guide. See this learn guide though:
https://learn.adafruit.com/customizing-usb-devices-in-circuitpython/circuitpy-midi-serial#usb-serial-console-repl-and-data-3096590
This is a guide that uses usb_cdc.data for "Remote Procedure Calls" from a Macropad to a host:
https://learn.adafruit.com/macropad-remote-procedure-calls-over-usb-to-control-home-assistant
And here is some minimal sample code I wrote to show how to exchange data between a PC and a Circuitpython board.
https://github.com/NeraSamples/circuitpython_usb_serial

If you need help with the implementation, please ask on the forums or the discord server, as support issues are better addressed there.

bernard01, 2025-05-09:
> Yes it is [supported], on EVERY raspberry pi board, since this is unrelated to the board, only to the chip's USB support.
Not on the Pico.

If I do in boot.py: 
`usb_cdc.enable(console=True, data=True)`

and nothing else

then I lock myself out of the console and I don't even see safe mode. The CIRCUITPY drive disappears and I have to nuke flash the board.

My request is about the Pico. The docs say it is supported but so far I haven't seen anyone report success with it.

deshipu, 2025-05-09:
How are you accessing the console? Are you selecting the correct COM port? When you enable the data CDC channel, there will be two COM ports on the pico, make sure your console is connecting to the right one.

bernard01, 2025-05-09:
> > The docs say it is supported but so far I haven't seen anyone report success with it.
> 
> I have done this before and it was fine. I don't believe you've specified your CircuitPython version in this report.

That is all I needed to hear. If you say that it worked for you with the Pico, then it must be my operating system. I tried the latest CP versions so that can't be the problem.

I will re-install Windows and then see whether I get 2 comm ports. I deleted the USB device multiple times. Every time it installs itself again it gets only 1 com port. It might be an issue with my bluetooth dongle or whatever. If that does not help I will try on a different computer. I am not ready yet for a compromise with a UART serial port yet.

Thank you all for your help so far.

bernard01, 2025-05-09:
I have downloaded and installed a fresh copy of Windows 10 on the same computer. No other competing devices. See device manager screen shot.

When I plug in the Pico, then Windows installs the drivers and as before, I get only 1 COM port, This time is easier than before because there is no other hardware with competing serial ports over USB. From my perspective, the CircuitPython software exposes only one serial port.

On the CIRCUITPY drive I installed adafruit-circuitpython-raspberry_pi_pico-en_US-10.0.0-alpha.4.uf2 which is the latest I could get. No boot.py. code.py has only Hello World in it.

According our expectations, this should be seen as a bug because we seem to agree that 2 COM ports should be available.

[image]

todbot, 2025-05-10:
> According our expectations, this should be seen as a bug because we seem to agree that 2 COM ports should be available.

Not quite. By default, you only get one USB CDC ("COM") port. This goes to the CircuitPython REPL.
If you want a second CDC port for non-REPL data transmission, then you must do:
```py
import usb_cdc
usb_cdc.enable(console=True, data=True)
```
in `boot.py` and power-cycle / hard reset the board to re-execute `boot.py`. 
The [default usb_cdc setting](https://docs.circuitpython.org/en/latest/shared-bindings/usb_cdc/index.html#usb_cdc.enable) is `usb_cdc.enable(console=True, data=False)`.

bernard01, 2025-05-10:
> import usb_cdc
> usb_cdc.enable(console=True, data=True)

You are right. This time it works. boot.py works as advertised. An additional port COM6 is created and the CIRCUITPY drive is still alive.

Many thanks.

todbot, 2025-05-10:
Awesome! If you feel this issue is resolved, you can go ahead and close it. Thanks!

closed 2025-05-10 by bernard01