adafruit/circuitpython #10342: Configure sdcardio automount within settings.toml

relic-se, 2025-05-13:
The new automount feature of `sdcardio` added in https://github.com/adafruit/circuitpython/pull/10129 is great for devices which host an SD card connector directly on board. This feature is currently unavailable on devices which have an SD card added on via a breakout or some other daughter board. In my case, I am using a Raspberry Pi Pico 2W with a Camera PiCowbell.

I suggest that new constants be added to `settings.toml` to define the pins, SPI device, and other properties used by this feature to allow it to work on devices which don't support it by default.

If it isn't possible to support this feature due to compile-time constraints and flash size, an alternative would be to add additional boards for well-supported addon boards such as the PiCowbell line. This change would definitely open the gates for many more combinations of boards and addon boards, so it wouldn't be ideal.

tannewt, 2025-05-14:
I don't really want to do this because it is up to code.py to set things up for things off the board. Setting up the sdcard there *should* still make it available to the host via USB though.

relic-se, 2025-05-14:
@tannewt I think that is my end goal here, to allow the second drive to be presented by the device to directly access the SD card. In my testing with 10.0.0-alpha.4, this hasn't been the case. Does the sd card configuration need to be defined within `boot.py` or has the implementation not been completed at this point?

tannewt, 2025-05-14:
You *must* use native sdcardio for it to work I believe. What is your `code.py`?

tannewt, 2025-05-14:
Looks like I have a check to ensure it isn't heap allocated. Try removing the `gc_nbytes()` check:
https://github.com/adafruit/circuitpython/blob/bbefbd6ce4986f66f12c87350e1610e8bc1318aa/supervisor/shared/usb/usb_msc_flash.c#L143-L144

relic-se, 2025-05-15:
@tannewt Huzzah! Removing the `gc_nbytes(sdcard) == 0` does in fact fix the situation and works for both examples even with removal and re-insertion. Awesome!

Disregard my initial settings.toml request. This is exactly what I was hoping for.

I'll make a quick PR for it and tie this issue to it to close as soon as it is merged. Thanks!

dhalbert, 2025-05-15:
Fixed by #10351

closed 2025-05-15 by dhalbert