adafruit/circuitpython #10345: Wiznet5K ethernet + requests using SSL causes hard crash with a pystack exhausted runtime error

dsmith9000, 2025-05-14:
### CircuitPython version and board name

```python
Circuitpython:
Version 9.2.7

Hardware:
1) Adafruit ESP32-S3 feather (4MB FLASH / 2MB PSRAM) + Adafruit Ethernet Featherwing
2) RP2040 PicoW + Adafruit Ethernet Featherwing (with minor modifications to the example code to address the differences between this platform and the ESP32-S3 platform)
```

### Code/REPL

```python
# Experiment with ethernet/ssl/tls and Circuitpython.
# Hardware: Adafruit ESP32-S3 4MB FLASH/2MB PSRAM + Adafruit Ethernet Featherwing
# Circuitpython v9.2.7
# 4/14/2025

import adafruit_connection_manager
import adafruit_requests
from adafruit_wiznet5k.adafruit_wiznet5k import WIZNET5K
from adafruit_wiznet5k.adafruit_wiznet5k_socketpool import SocketPool
from adafruit_wiznet5k.adafruit_wiznet5k_dns import DNS
import array
import board
import busio
import digitalio
import gc
import ssl
import sys

# Connection method options
METHOD_REQUESTS = "requests"        # Use 'requests' to make connection
METHOD_SOCKETS = "sockets"          # Use 'sockets' to make connection

# Select method to use
method_to_use = METHOD_REQUESTS

# Configuration
W5500_CS_PIN = board.D10

def print_ipaddress(ip_address: bytes|tuple(int)):
    addy = [str(octet) for octet in ip_address]
    print(".".join(addy))

def print_ifconfig(ifconfig):
    print("ifconfig: ", end="")
    for ip in ifconfig:
        print(".".join(f"{byte}" for byte in ip), " ", end="")
    print("\n")

def connect_using_https_with_requests(eth):
    print("<<HTTPS USING REQUESTS>>")
        
    # Initialize a requests session
    pool = adafruit_connection_manager.get_radio_socketpool(eth)
    ssl_context = adafruit_connection_manager.get_radio_ssl_context(eth)
    requests = adafruit_requests.Session(pool, ssl_context)

    gc.collect()
    try:
        r = requests.get("https://www.google.com", stream=True)     # RuntimeError('pystack exhausted') occurs here

    except RuntimeError as ex:
        print("Error :", repr(ex))
        sys.exit()
    
    print("Bytes received: ", len(r.text))
    print("First 100 bytes of response:\n", r.text[:100])
[... 230 lines cut ...]
*** Get socket.
Allocated socket # 0
W5K socket connect, protocol=2, port=53, ip=192.168.100.1
*** Opening socket 0
* Opening W5k Socket, protocol=2
Error : RuntimeError('pystack exhausted',)

Code done running.

### Description

Bug Report:
Hard crash due to RuntimeError("pystack exhausted") when using 'adafruit_requests' to connect to an https website (google) on 2 different hardware platforms using Wiznet5K Ethernet phy chips.

Hardware:
1) Adafruit ESP32-S3 feather (4MB FLASH / 2MB PSRAM) + Adafruit Ethernet Featherwing
2) RP2040 PicoW + Adafruit Ethernet Featherwing (with minor modifications to the example code to address the differences between this platform and the ESP32-S3 platform)

Circuitpython:
Version 9.2.7

Description:
I've been attempting to evaluate the use of hardwired Ethernet via Wiznet5K chips and Circuitpython. When using the adafruit_requests package and attempting to connect to an https site (google), I get a hard crash due to a RuntimeError("pystack exhausted") on 2 different hardware platforms. If I don't use adafruit_requests and instead just use the sockets module along with the ssl module, I am able to successfully connect and receive the response.

I've attached my simple POC code that reproduces this behavior.  As listed, the code runs the "https with adafruit_requests" test and demonstrates the failure mode.  To see the successful use of socket + ssl to connect, change the line 'method_to_use = METHOD_REQUESTS' to 'method_to_use = METHOD_SOCKETS'.

anecdata, 2025-05-14:
pystack default is getting increased in CP 10. For now, you could try in `settings.toml`:

`CIRCUITPY_PYSTACK_SIZE=2048`

dsmith9000, 2025-05-14:
@anecdata Thank you for this suggestion!  That indeed resolves my problem and allows the get to successfully receive a response.