adafruit/circuitpython #10353: Deep Sleep not available from REPL

eightycc, 2025-05-16:
Attempting deep sleep from REPL by `alarm.exit_and_deep_sleep_until_alarms` exits immediately without sleeping. This is because the REPL implementation does not exit its process loop when a `PYEXEC_DEEP_SLEEP` is thrown.

This should be fixed or documented as a limitation.

bablokb, 2025-05-16:
I think this is documented, even in bold, see https://docs.circuitpython.org/en/latest/shared-bindings/alarm/index.html

eightycc, 2025-05-16:
> I think this is documented, even in bold, see https://docs.circuitpython.org/en/latest/shared-bindings/alarm/index.html

I'm not seeing it, or I'm misunderstanding the text.

anecdata, 2025-05-16:
Further down under `exit_and_deep_sleep_until_alarms` is:

> If CircuitPython is connected to a host computer via USB or BLE the first time a deep sleep is requested, the connection will be maintained and the system will not go into deep sleep.

(But TIL:

> If CircuitPython goes into a true deep sleep, and USB or BLE is reconnected, the next deep sleep will still be a true deep sleep. You must do a hard reset or power-cycle to exit a true deep sleep loop.

)