adafruit/circuitpython #10358: Update carlosperate/arm-none-eabi-gcc-action after it updates its use of cache

dhalbert, 2025-05-17:
We are getting a lot of `Failed to restore: Cache service responded with 422` warnings during GitHub Actions builds. This is apparently due to `carlosperate/arm-none-eabi-gcc-action` not updating its use of `actions/cache` yet to `actions/cache@v4`. @carlosperate is planning to fix this: https://github.com/carlosperate/arm-none-eabi-gcc-action/issues/64#issuecomment-2829760318

carlosperate, 2025-05-22:
The latest release updates de action cache lib and fixes the warnings. Thanks for the reminder @dhalbert !

dhalbert, 2025-05-22:
Fixed by https://github.com/carlosperate/arm-none-eabi-gcc-action/releases/tag/v1.10.1, which we should get automatically, because it's still `@v1`.

closed 2025-05-22 by dhalbert