adafruit/circuitpython #10360: USB Libraries Missing from Teensy 4.0 CircuitPython Builds

Ahsoka, 2025-05-19:
### CircuitPython version and board name

```python
Teensy 4.0; all versions of CircuitPython
```

### Code/REPL

```python
pass
```

### Behavior

Hello,

I am trying to my Teensy 4.0 with USB libraries and I noticed that none of the USB libraries are supported for the Teensy 4.0, however, they are supported for the Teensy 4.1.  As far as I am aware the Teensy 4.0 and Teensy 4.1 have the same USB functionality.  Would it be possible to add the USB libraries to the Teensy 4.0?

todbot, 2025-05-19:
Can you describe exactly which libraries are missing?  I tried the below in the REPL on a Teensy 4.0 and seems like they're all there:
```py
Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 9.2.7 on 2025-04-01; Teensy 4.0 with IMXRT1062DVJ6A
>>> import usb_hid, usb_cdc, usb_midi
>>> 
```

If you're talking about the libraries `adafruit_hid` or `adafruit_midi`, those you copy yourself or use `circup` to install. The official library bundles live at https://circuitpython.org/libraries Or you can grab a whole keyboard or mouse project bundle with libraries here: https://learn.adafruit.com/circuitpython-essentials/circuitpython-hid-keyboard-and-mouse  

To verify USB works on a Teensy 4.0, I saved this as `code.py` and it works as expected, moving the mouse cursor:
```py
import time
import usb_hid
from adafruit_hid.mouse import Mouse
mouse = Mouse(usb_hid.devices)
while True: 
    mouse.move(50,50)
    time.sleep(0.5)
    mouse.move(-50, -50)
    time.sleep(0.5)
```

Neradoc, 2025-05-20:
From what I can see, the USB OTG2 pins that could be used on the teensy 4.0 are the two pads on the bottom of the board (that despite their placement are actually not the same as the USB connector. So, it looks like it should be possible to enable USB host via those pins, but I don't know if replicating the configuration from the Teensy 4.1 will work.
Also, I don't want to solder on the bottom pads of my teensy for testing and I don't know if it needs additional components.

I've made a commit to test it, you can try the hex from the build (teensy40 zip artifact at the bottom of the page).
https://github.com/Neradoc/circuitpython/actions/runs/15138187456

This doesn't raise any exception, but that's all I can test (it prints nothing as expected with nothing connected to it)
```py
import board
import usb
import usb_host

port = usb_host.Port(board.USB_HOST_DP, board.USB_HOST_DM)

for device in usb.core.find(find_all=True):
    print(device)
```

todbot, 2025-05-21:
I found my Teensy 4.0 with a USB jack soldered to it and tested this out.  Yep, works!  

It works with a USB hub too.  But if you unplug the hub and then go back to the device plugged in directly, it's not recognized. Is this a known bug?  I've not played with USB host stuff recently in CirPy.

Anyway, here's what my Teensy 4.0 USB host wiring from six years ago looks like: 

[image]

Also, since `usb.core.Device` doesn't seem to contain a useful `__str__()` or `__repr___()`, my variation of the above looks like this:

```py
import time, board
import usb, usb_host

port = usb_host.Port(board.USB_HOST_DP, board.USB_HOST_DM)

while True:
    for device in usb.core.find(find_all=True):
        print(device.product, device.manufacturer, hex(device.idProduct), hex(device.idVendor), time.monotonic())
    time.sleep(0.5)
```