adafruit/circuitpython #10362: i2ctarget error with missing deinit for request

Neradoc, 2025-05-19:
Since #10040 (9.2.5 and after) this happens with the i2ctarget example:

```py
code starting...
9.789: INFO - read request to address '0x40'
Traceback (most recent call last):
  File "code.py", line 38, in <module>
AttributeError: 'I2CTargetRequest' object has no attribute 'deinit'
```

This is apparently because the new default `__exit__` requires the presence of `deinit` in the object's dictionary, and `I2CTargetRequest` doesn't have one. It has a `close()` though, (which incidentally doesn't have a docstring).

Should it have a `deinit()` that calls `close()` and also maybe remove the `close()` ?

dhalbert, 2025-05-19:
I think this is the following omission: if you look in the `dict_table` for, say `I2CTarget`, you'll see:
```c
    { MP_ROM_QSTR(MP_QSTR_deinit), MP_ROM_PTR(&i2ctarget_i2c_target_deinit_obj) },
    { MP_ROM_QSTR(MP_QSTR___del__), MP_ROM_PTR(&i2ctarget_i2c_target_deinit_obj) },

```
The presence of the `__del__` entry is true for all objects with a finaliser, which should be true for enter/exit objects. In `i2ctarget_i2c_target_make_new()`, the object created is created with `mp_obj_malloc_with_finaliser()`.

I think the same pattern should be used for `I2CTargetRequest`: a `deinit()`, use `mp_obj_malloc_with_finaliser()`, and add the `deinit` and `___del__` to the method table.

Is there any example code that uses the `close`??

dhalbert, 2025-05-20:
I looked at the original PR, #1064, which mostly survives as is in the current code. I think the best thing to do here is to remove a user-accessible `close()` method, add `deinit()`, and have `__exit__()` call the common-hal target `close()` and then do a `deinit()`. I am working on a PR.

jbrelwof, 2025-05-21:
I'm curious - beyond searching the Adafruit and community bundles, is there any standard(-ish) process to search for method usage in a wider body of existing CircuitPython code?    Obviously it needs to be something a bit more intelligent than `simply grep close ...`, but there are a variety of options for finding references to python method use that (try to and usually succeed at least partially) filter on the type, i.e. find invocations of **close()** on an **I2CTargetRequest**.   But I'm not sure _what_ you would search - searching github for CircuitPython currently returns nearly 4K repos, add MicroPython and you're pushing 17K.  Is there any existing list of more widely used (but not in the Adafruit/community bundles) modules for testing / API deprecation checks?

dhalbert, 2025-05-21:
I search GitHub in general, both in the Adafruit org and elsewhere, and often do a general websearch, if I can make it specific enough.

PR #10366 (2025-05-21): i2ctarget: Add deinit() to I2CTargetRequest; remove close()
- Fixes #10362.

Removes visible `I2CTargetRequest.close()` method. Adds `I2CTargetRequest.deinit()`. Does `close()` and `deinit()` in `I2CTargetRequest.__exit__()`. Makes `I2CTargetRequest` have a finalizer: `__del__()` calls `deinit()`.

I tried testing with the more extensive example in the `I2CTarget` documentation, which tries to write and then read a value to an I2C register. It did not work, but I could not get that example to work with 8.2.9 or 9.2.3 either.

@Neradoc if you or someone else would like to test this with some pr [... cut]
closed 2025-05-21 by dhalbert