adafruit/circuitpython #10365: How to build nrf52840 with 1.8V QSPI Flash?

CaptainJackey, 2025-05-21:
Take feather_nrf52840_express as an example, it uses W25Q16JVxQ with 3.3V flash. Now I need to use a 1.8V power system, equipped with W25Q16JWxQ with 1.8V flash. What code should I modify to achieve this?

dhalbert, 2025-05-21:
You can change the GPIO voltage by setting bits in the UICR (search for that in the datasheet, etc.). Is everything 1.8V? I am not sure if there is brownout detection setup in the bootloader, in https://github.com/adafruit/Adafruit_nRF52_Bootloader. Look to see.

https://devzone.nordicsemi.com/ is the Nordic forum area, which may have answers for other questions you may have.

CaptainJackey, 2025-05-22:
> You can change the GPIO voltage by setting bits in the UICR (search for that in the datasheet, etc.). Is everything 1.8V? I am not sure if there is brownout detection setup in the bootloader, in [...]

Yes, everything 1.8V. You mean the REGOUT0? It seems only work in **high voltage mode**, but my circuit is not this mode , It is powered by an external 1.8V connect VDD and VDDH. 

[image]

dhalbert, 2025-05-22:
Then it sounds like you really have to do nothing except change the device ID for the different flash. The datasheets look pretty much identical except for the Device ID and the voltages (see screenshow below). The flash chips are described in https://github.com/adafruit/nvm.toml, which is a submodule. You could just edit that locally to add the new chip in the `winbond` directory, or fork it, or submit a PR to add the chip. 

In `ports/nordic/supervisor/port.c`, in `port_init()`, the brownout detection is set to 2.7v, and you'll need to change that and maybe do brownout detection in a different way. When I read the datasheet section it seemed like the POFCON threshold values might be done different for low voltages and the different voltage configuration.

`analogio` assumes 3.3v, and you'll want to change `common_hal_analogio_analogin_get_reference_voltage()`. There may be other things to change in `analogio`.

[image]

dhalbert, 2025-05-22:
https://github.com/adafruit/circuitpython/issues/6501 may possibly be an issue for you? Are you using USB?

closed 2025-05-26 by CaptainJackey