adafruit/circuitpython #10368: PIO write delayed by i2c display update

dmcomm, 2025-05-23:
### CircuitPython version and board name

```python
Adafruit CircuitPython 9.2.7 on 2025-04-01; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
import array
import time

import board
import digitalio
import displayio
import rp2pio
import busio
import i2cdisplaybus
import terminalio
import adafruit_displayio_ssd1306
import adafruit_pioasm
from adafruit_display_text.bitmap_label import Label

program = adafruit_pioasm.assemble("""
    pull
    mov osr ~ osr
    set pins 1 [8]
    set x 7
loop:
    out pins 1 [8]
    jmp x-- loop
    set pins 0 [7]
""")

displayio.release_displays()
i2c = busio.I2C(board.GP5, board.GP4)
display_bus = i2cdisplaybus.I2CDisplayBus(i2c, device_address=0x3c)
display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=64)

probe = digitalio.DigitalInOut(board.GP0)
probe.switch_to_output(value=True)

output = rp2pio.StateMachine(
    program,
    frequency=25_000,
    first_out_pin=board.GP16,
    first_set_pin=board.GP16,
)

groups = []
for text in ["Hello", "World"]:
    group = displayio.Group()
    label = Label(terminalio.FONT, text=text, color=0xFFFFFF, x=10, y=10)
    group.append(label)
    groups.append(group)

def send(data):
    start_time = time.monotonic()
    probe.value = True
    output.write(data)
    probe.value = False
    print((time.monotonic() - start_time) * 1000)

bytes_to_send = bytes([0] * 15)
i = 0
while True:
    display.root_group = groups[i]
    send(bytes_to_send)
    time.sleep(5)
    i = 1 - i
```

### Behavior

Expecting write to take 60ms with these values, which it does if we remove the display update, but it intermittently reports over 100ms. Scope trace shows the transmission is finished before the delay. Setting frequency to higher numbers gives more extreme results but is harder to see the details.

As above:
[image]

Without display update:
[image]

With display update, frequency=200_000:
[image]

Reverting c5c7d11 gives a similar amount of delay, but in the middle instead:
[image]

dmcomm, 2025-05-23:
Requesting a 5ms sleep after the display update really sleeps for 5ms and has no effect on the problem. Requesting a 20ms sleep actually sleeps for 100ms but resolves the problem.

dhalbert, 2025-05-23:
Could you try 10.0.0-alpha.6 and see if it's different? The sleep behavior you just mentioned may be different. This seems like two memory-heavy operations competing.

dmcomm, 2025-05-23:
The delay is still present on 10.0.0-alpha.6

eightycc, 2025-05-23:
@dmcomm Thanks for the issue!

With #8700, `compute_fifo_depth()` returns the PIO FIFO depth in 32-bit words, but `_transfer()` treats it as bytes. This causes `_transfer()` to select DMA for the transfer for some transfer sizes smaller than the PIO FIFO size. While this isn't a breaking bug, it does shed some light on the results you're seeing.

At the conclusion of `_transfer()`, a while loop waits for the PIO's stall bit to go off before concluding the operation: https://github.com/adafruit/circuitpython/blob/ef9e14bb39909e6fc8259349c996e2135dfe046c/ports/raspberrypi/common-hal/rp2pio/StateMachine.c#L998-L1008

This could be where the delay is occurring.

dmcomm, 2025-05-23:
I was interested to know whether removing `RUN_BACKGROUND_TASKS` there would prevent the delay, but it doesn't seem to.

And from what I wrote in the OP, I thought the workaround would be easy, but in my real program something else seems to be having this effect too.

eightycc, 2025-05-23:
@dmcomm Please try adding `wait_for_tx_stall=False` to the argument list for the state machine constructor.

dmcomm, 2025-05-23:
Returning before "Wait for the state machine to finish" doesn't resolve the delay, so the delay isn't in that section, even though according to the scope the transmission is finished. The times when the delay does not occur, it returns after 6 bytes with 9 remaining, which is 8 in the FIFO and 1 in progress as expected.

mikeysklar, 2026-11-09:
The stall happens because the display refresh runs inside the `write()` call.

A workaround is to set `display.auto_refresh = False` and call `display.refresh()` between writes.

Tested on a Feather RP2040 with an SSD1306 over I2C, on current main.

| Write | Stock | With workaround |
|---|---|---|
| 15 bytes, n=600 | 113 ms median, all over 65 ms | 60.2 ms, 0 over |
| 4 bytes, n=150 | not tested | 16.2 ms, 0 over |

Each `refresh()` takes about 103 ms, so the time moves out of the write rather than going away.

mikeysklar, 2026-11-14:
Your scope traces show the PIO output is on time. 

Only the return from `write()` is late, because the display refresh runs inside it.

There are two reasonable work arounds to reduce / remove the blocking delay.

| Setup | Added to a `write()` that overlaps a refresh |
|---|---|
| Today, 100 kHz I2C | about 50 to 100 ms |
| 400 kHz I2C | about 25 ms (estimate, not measured) |
| `auto_refresh = False` plus `display.refresh()` between writes | 0 |
| Non-blocking I2C display send (not built) | close to 0 |

The last option for non-blocking I2C isn't out of the question.  #10856 is exactly that, but it's likely going to cost 2KB of RAM and that is hard to justify on the RP2040.

tannewt, 2026-11-16:
I'll mark it as a duplicate so the issues are linked.

closed 2026-11-16 by tannewt