adafruit/circuitpython #10390: espressif: CIRCUITPY drive missing in 10.0.0-alpha.6

Sola85, 2025-05-31:
### CircuitPython version and board name

```python
Waveshare ESP32-S3-Zero 10.0.0-alpha.6 with Windows 11
```

### Code/REPL

```python
Doesn't matter
```

### Behavior

After erasing flash and flashing 10.0.0-alpha.6, no CIRCUITPY drive appears after plugging in.
`main.py` seems to run, but I can't interact with the board via usb serial.
I see the same behaviour on main at https://github.com/adafruit/circuitpython/commit/002881826a04669abaaa4f2d5220379f786cb159.

Flashing 9.2.8 again and everything works again.

Sola85, 2025-06-02:
Hadn't realized this before, but when the CIRCUITPY drive does show up after a few minutes in the problematic configuration, it is read-only.

I tried removing `code.py`, `main.py` and `settings.toml` and also tried replacing them with empty files (`settings.toml` was empty all along), but this doesn't change anything. These changes to the files were only possible on linux, due to the above problem. 

I also tried a second Windows 11 machine, but this behaves the same as the first: No/Very late Circuitpy drive on 10.0.0-alpha.6 and working circuitpy drive on 9.2.8.

dhalbert, 2025-07-01:
I have reproduced this problem with a QT Py ESP32-S3 4MB/2MB, running its own firmware, and also running the Waveshare Zero 4/2 firmware. As you reported, it works fine on Ubuntu, but on Windows 11, CIRCUITPY takes a long time to show up, and is read-only.

Very interestingly, a QT Py ESP32-S3 8MB works fine on Windows 11. CIRCUITPY appears right away, and is read/write. So it may have to do with the presence of the PSRAM.

@Sola85 What is the flash/PSRAM configuration of the Generic ESP32-S3-MINI-1 you tried?

Sola85, 2025-07-02:
The working board was using the ESP32-S3-MINI-1-N8, so 8MB flash and no PSRAM. 
The Waveshare board has 4MB flash and 2MB PSRAM.
So I believe this matches your findings.

dhalbert, 2025-07-02:
I disabled PSRAM several different ways on a 4MB build, and it still had problems. :( I am wondering if it has to do with the size of CIRCUITPY. I'll try that next. There are very few other differences between the two builds.

dhalbert, 2025-07-02:
I made the 8MB QT Py build CIRCUITPY be the same size as the 4/2 build: 960kB. Works on Windows 11 fine. So it appears to be yet something else.

dhalbert, 2025-07-03:
The crucial difference is that the 8MB build is compiled mostly with `-O2`; the 4MB build is mostly `-Os` to fit in the smaller firmware partition. If I compile the 8MB build with `-Os`, it starts having the same problem.

Now to figure out why. This could be due to something uninitialized or something that should be `volatile` but is not.

Yup, a bisect may help certainly help here, and that is probably the next thing.

dhalbert, 2025-07-04:
Setting `CIRCUITPY_SDCARDIO = 0`  causes the problem to go away; turns off some things added in #10129, and reduces the LUN count back to 1. It's still weird that it takes `-Os` to have the problem appear. On a whim, I tried making the `ejected[]` etc .arrays in `shared/supervisor/usb_msc_flash.c` be `volatile`: no help.

I am posting these comments just to help keep track of what was tried.