adafruit/circuitpython #10392: LILYGO T-Dongle S3 fails in bootloader

WillB97, 2025-05-31:
### CircuitPython version and board name

```python
adafruit-circuitpython-lilygo_tdongle_s3-en_GB-9.2.5.bin
```

### Code/REPL

```python
The REPL never reaches a point to execute code and the CIRCUITPY drive never appears.
```

### Behavior

After flashing the circuitpython binary, no USB drive appears and the REPL prints the following repeatedly.
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378b81
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x18c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xa7c
load:0x403cb700,len:0x2740
entry 0x403c8854
```

### Additional information

I have managed to get MicroPython running on the dongle using the firmware that LILYGO provides in [their repository](https://github.com/Xinyuan-LilyGO/T-Dongle-S3/tree/main/example/Micropython/firmware).

In https://github.com/adafruit/circuitpython/pull/10065/files#diff-75938e484e23c93e62559507d7c14b81ecfebf40557f4cd03aa4dcad99ea56e3R12 it looks like the dongle expects to have PSRAM but LILYGO's schematics don't look to include any PSRAM.

lewisxhe, 2025-07-08:
Yes, this board does not have PSRAM. Turning on PSRAM will get stuck at startup.

lewisxhe, 2025-07-08:
Hi dhalbert
I am a LilyGo employee and I can confirm that this board does not have PSRAM.

closed 2025-07-08 by dhalbert

deisterhold, 2025-07-08:
@dhalbert I'll defer to @lewisxhe's expertise. My apologies for accidentally including that config.

I've tested the latest build and confirmed that the REPL is accessible.

[image]

lewisxhe, 2025-07-09:
[image]
Can confirm that it can be accessed correctly