adafruit/circuitpython #10428: Crash

kenidowa, 2025-06-17:
### CircuitPython version and board name

```python
Adafruit CircuitPython 9.2.4 on 2025-01-29; Adafruit QT Py ESP32 PICO with ESP32
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2023 Michał Pokusa
#
# SPDX-License-Identifier: Unlicense

from asyncio import create_task, gather, run
from asyncio import sleep as async_sleep

import board
import microcontroller
import neopixel
import socketpool
import wifi

from adafruit_httpserver import GET, Request, Response, Server, Websocket

#******************
print(dir(microcontroller))
print(dir(microcontroller.Processor))
print(dir(microcontroller.cpu))
print(microcontroller.cpu.frequency)
print(microcontroller.cpu.voltage)

pool = socketpool.SocketPool(wifi.radio)
server = Server(pool, debug=True)

pixel = neopixel.NeoPixel(board.NEOPIXEL, 1)

websocket: Websocket = None

HTML_TEMPLATE = """
<html lang="en">
    <head>
        <title>Websocket Client</title>
    </head>
    <body>
        <p>Cycle Count: <strong>-</strong></p>
        <p>NeoPixel Color: <input type="color"></p>
        <script>
            const cpuTemp = document.querySelector('strong');
            const colorPicker = document.querySelector('input[type="color"]');

            let ws = new WebSocket('ws://' + location.host + '/connect-websocket');

            ws.onopen = () => console.log('WebSocket connection opened');
            ws.onclose = () => console.log('WebSocket connection closed');
            ws.onmessage = event => cpuTemp.textContent = event.data;
            ws.onerror = error => cpuTemp.textContent = error;

            colorPicker.oninput = debounce(() => ws.send(colorPicker.value), 200);

            function debounce(callback, delay = 1000) {
                let timeout
                return (...args) => {
                    clearTimeout(timeout)
                    timeout = setTimeout(() => {
                    callback(...args)
                  }, delay)
                }
            }
        </script>
    </body>
</html>
"""

@server.route("/client", GET)
def client(request: Request):
    return Response(request, HTML_TEMPLATE, content_type="text/html")

@server.route("/connect-websocket", GET)
def connect_client(request: Request):
    global websocket

    if websocket is not None:
        websocket.close()  # Close any existing connection

    websocket = Websocket(request)

    return websocket

server.start(str(wifi.radio.ipv4_address))

async def handle_http_requests():
    while True:
        server.poll()

        await async_sleep(0)

async def handle_websocket_requests():
    while True:
        if websocket is not None:
            if (data := websocket.receive(fail_silently=True)) is not None:
                r, g, b = int(data[1:3], 16), int(data[3:5], 16), int(data[5:7], 16)
                pixel.fill((r, g, b))

        await async_sleep(0)

async def send_websocket_messages(cpu_temp):
    while True:
        if websocket is not None:
            #cpu_temp = round(microcontroller.cpu.frequency, 0)
            cpu_temp +=1
            #cpu_temp = round(0.5, 2)
            websocket.send_message(str(cpu_temp), fail_silently=True)

        await async_sleep(1)

async def main():
    await gather(
        create_task(handle_http_requests()),
        create_task(handle_websocket_requests()),
        create_task(send_websocket_messages(0)),
    )

run(main())

ruit CircuitPython 9.2.4 on 2025-01-29; Adafruit QT Py ESP32 PICO with ESP32
```

### Behavior

OUTPUT from Serial port:
soft reboot

code.py output:
['__class__', '__name__', 'Pin', 'Processor', 'ResetReason', 'RunMode', '__dict__', 'cpu', 'delay_us', 'disable_interrupts', 'enable_interrupts', 'nvm', 'on_next_reset', 'pin', 'reset', 'watchdog']
['__class__', '__name__', '__bases__', '__dict__', 'frequency', 'reset_reason', 'temperature', 'uid', 'voltage']
['__class__', 'frequency', 'reset_reason', 'temperature', 'uid', 'voltage']
240000000
None
Started development server on http://192.168.4.235:5000
192.168.5.13 -- "GET /client" 438 -- "200 OK" 1244 -- 31ms
192.168.5.13 -- "GET /connect-websocket" 511 -- "101 Switching Protocols" 149 -- 31ms
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:396
ho 0 tail 12 room 4
load:0x40078000,len:13892
load:0x40080400,len:4
load:0x40080404,len:3160
entry 0x40080558
Serial console setup

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

### Description

Run this code on ESP32-PICO ran for about 2 days and then crashed hard.  This code comes from
This link: https://learn.adafruit.com/networking-in-circuitpython/http-server-examples

anecdata, 2025-06-17:
Can you re-test with the latest CircuitPython 10 alpha? There was a fix for HTTP server safe mode faults in <https://github.com/adafruit/circuitpython/pull/10328>.

kenidowa, 2025-06-17:
Ok reloaded with 10.0.0-alpha.6 
got this failure:

soft reboot

code.py output:
['__class__', '__name__', 'Pin', 'Processor', 'ResetReason', 'RunMode', '__dict__', 'cpu', 'delay_us', 'disable_interrupts', 'enable_interrupts', 'nvm', 'on_next_reset', 'pin', 'reset', 'watchdog']
['__class__', '__name__', '__bases__', '__dict__', 'frequency', 'reset_reason', 'temperature', 'uid', 'voltage']
['__class__', 'frequency', 'reset_reason', 'temperature', 'uid', 'voltage']
240000000
None
Started development server on http://192.168.4.235:5000
Traceback (most recent call last):
  File "code.py", line 124, in <module>
  File "asyncio/core.py", line 317, in run
  File "asyncio/core.py", line 228, in create_task
AttributeError: 'TaskQueue' object has no attribute 'push_head'

Code done running.

Adafruit CircuitPython 10.0.0-alpha.6 on 2025-05-17; Adafruit QT Py ESP32 PICO with ESP32
>>>

anecdata, 2025-06-17:
Try updating the `asyncio` library. There was a change in April that affected this:

https://github.com/adafruit/Adafruit_CircuitPython_asyncio/releases/tag/3.0.0

Best to update all libraries since you're now using Circuitpython 10:
https://circuitpython.org/libraries
Use libraries from the Bundle for CircuitPython 10.x

kenidowa, 2025-06-17:
I updated all the libraries with 10.x library bundle and asyncio was in that.[

[VERSIONS.txt](https://github.com/user-attachments/files/20785347/VERSIONS.txt)

](url)

anecdata, 2025-06-18:
That solved it then? I loaded up an ESP32-S3 QT Py with the code above, 10.0-alpha.7, and the needed 10.x libraries, and it seems to run fine.

kenidowa, 2025-06-18:
No, 10.x Libraries is what I ran it when it failed the the TaskQueue issue with push_head.  When I do a dir(asyncio.TaskQueue) I get:
['__class__', '__name__', 'pop', 'remove', '__bases__', '__dict__', 'peek', 'push']

anecdata, 2025-06-18:
What's your `asyncio` version:
```
>>> asyncio.__version__
'3.0.1'
```

Can you list or show a list of the files at the root level of CIRCUITPY, and also in the /lib folder? 

I think there could be some conflict with library files. `.py` will always take precedence over `.mpy`, and libraries at the root will always take precedence over libraries in /lib

kenidowa, 2025-06-18:
This is crude with misc html around so you'll have to overlook that. I was having problem figuring how to display it here.

/</
Type | Size | Path | Modified |  
-- | -- | -- | -- | --
📁 | 0 B | lib | 12/31/1999, 5:00:04 PM | ✏️ Rename | 🗑️ Delete |  
📁 | 0 B | sd | 12/31/1999, 5:00:02 PM | ✏️ Rename | 🗑️ Delete |  
📄 | 167 B | boot_out.txt | 12/31/1999, 5:00:08 PM | ✏️ Rename | 🗑️ Delete | 📝 Edit
📄 | 3.3 kB | code.py | 6/17/2025, 11:36:36 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit
📄 | 100 B | settings.toml | 6/17/2025, 9:46:28 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit

/lib/
Type | Size | Path | Modified |  
-- | -- | -- | -- | --
📁 |   | .. |   |   |   |  
📁 | 0 B | adafruit_httpserver | 6/17/2025, 4:23:50 PM | ✏️ Rename | 🗑️ Delete |  
📁 | 0 B | asyncio | 6/17/2025, 4:22:46 PM | ✏️ Rename | 🗑️ Delete |  
⬇️ | 694 B | adafruit_ticks.mpy | 6/14/2025, 5:12:36 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit
⬇️ | 1.3 kB | neopixel.mpy | 6/14/2025, 5:12:38 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit

/lib/adafruit_httpserver/
Type | Size | Path | Modified |  
-- | -- | -- | -- | --
📁 |   | .. |   |   |   |  
⬇️ | 1.6 kB | __init__.mpy | 6/14/2025, 5:12:36 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit
⬇️ | 849 B | authentication.mpy | 6/14/2025, 5:12:36 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit
⬇️ | 645 B | exceptions.mpy | 6/14/2025, 5:12:36 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit
⬇️ | 1.1 kB | headers.mpy | 6/14/2025, 5:12:36 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit
⬇️ | 1.9 kB | interfaces.mpy | 6/14/2025, 5:12:36 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit
⬇️ | 155 B | methods.mpy | 6/14/2025, 5:12:36 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit
⬇️ | 3.4 kB | mime_types.mpy | 6/14/2025, 5:12:36 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit
⬇️ | 3.8 kB | request.mpy | 6/14/2025, 5:12:36 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit
⬇️ | 6.7 kB | response.mpy | 6/14/2025, 5:12:36 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit
⬇️ | 1.7 kB | route.mpy | 6/14/2025, 5:12:36 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit
⬇️ | 6.1 kB | server.mpy | 6/14/2025, 5:12:36 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit
⬇️ | 1.2 kB | status.mpy | 6/14/2025, 5:12:36 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit

lib/asyncio/
Type | Size | Path | Modified |  
-- | -- | -- | -- | --
📁 |   | .. |   |   |   |  
⬇️ | 473 B | __init__.mpy | 12/3/2024, 5:11:08 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit
⬇️ | 3.2 kB | core.mpy | 12/3/2024, 5:11:08 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit
⬇️ | 647 B | event.mpy | 12/3/2024, 5:11:08 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit
⬇️ | 1.1 kB | funcs.mpy | 12/3/2024, 5:11:08 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit
⬇️ | 609 B | lock.mpy | 12/3/2024, 5:11:08 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit
⬇️ | 193 B | manifest.mpy | 12/3/2024, 5:11:08 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit
⬇️ | 1.9 kB | stream.mpy | 12/3/2024, 5:11:08 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit
⬇️ | 1.5 kB | task.mpy | 12/3/2024, 5:11:08 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit
⬇️ | 645 B | traceback.mpy | 12/3/2024, 5:11:08 AM | ✏️ Rename | 🗑️ Delete | 📝 Edit

anecdata, 2025-06-18:
Can you completely delete the asyncio folder, and copy over the asyncio folder fresh from the 10.x bundle (or use `circup`)?

It still seems there is an old version lingering.
``
File "asyncio/core.py", line 228, in create_task
AttributeError: 'TaskQueue' object has no attribute 'push_head'
``
This does not exist in asyncio version 3.

kenidowa, 2025-06-18:
An update from my end.  I reloaded asyncio with the lates library 20250618 10.x bundle.  It didn't crash and is behaving better.  It was a bit strange at startup.  It took ~2 mins to connect and the counter I added didn't seem to respond but then tried reconnecting twice and seems to be working correctly. I am going to let it run and see what happens.

kenidowa, 2025-06-18:
Started development server on http://192.168.4.235:5000
192.168.5.13 -- "GET /client" 438 -- "200 OK" 1236 -- 123506ms
192.168.5.13 -- "GET /client" 464 -- "200 OK" 1236 -- 13ms
192.168.5.13 -- "GET /client" 438 -- "200 OK" 1236 -- 13ms
192.168.5.13 -- "GET /connect-websocket" 511 -- "101 Switching Protocols" 149 -- 20ms
192.168.5.13 -- "GET /client" 438 -- "200 OK" 1236 -- 14ms
192.168.5.13 -- "GET /connect-websocket" 511 -- "101 Switching Protocols" 149 -- 21ms
192.168.5.13 -- "GET /client" 438 -- "200 OK" 1236 -- 23ms
192.168.5.13 -- "GET /connect-websocket" 511 -- "101 Switching Protocols" 149 -- 21ms
This is activity on serial port at startup.

anecdata, 2025-06-18:
Sounds like things are generally working now. I'm going to close this. The discussion has gone pretty far afield from the original issue. For support, the Adafruit Discord is a great place for community help to get things working. There's also the Adafruit forums. If it's something specific to the HTTP Server library, it's best to file a new issue there.

closed 2025-06-18 by anecdata