adafruit/circuitpython #10444: Remove pre 7.x compatibility from OnDiskBitmap?

FoamyGuy, 2025-06-27:
While looking into something else I found some libraries that have usages of pre 7.x OnDiskBitmap API that required opening the file before creating the ODB instance. I created this issue for the place I found it: https://github.com/adafruit/Adafruit_CircuitPython_SSD1680/issues/34 and haven't searched for more in other libraries / learn code yet but do intend to. 

In the ODB docs https://docs.circuitpython.org/en/latest/shared-bindings/displayio/index.html#displayio.OnDiskBitmap it mentions:

> Older versions of CircuitPython required a file opened in binary mode. CircuitPython 7.0 modified OnDiskBitmap so that it takes a filename instead, and opens the file internally. A future version of CircuitPython will remove the ability to pass in an opened file.

But we have not yet removed that backwards functionality. Do we want to do that with 10.x?

dhalbert, 2025-06-27:
This sounds fine to remove, as long as any projects are modified. It's certainly been deprecated long enough.

dhalbert, 2025-07-16:
I looked at this. It's easy to remove the ability to pass an open file. But internally, it just opens the file, and never closes it. There is currently no context manager support: no `deinit()` or `__del__()` to close the file. Ideally that should be done.

closed 2026-11-04 by dhalbert