adafruit/circuitpython #10448: Allow other fixed fonts for terminals besides terminalio.FONT

TheKitty, 2025-06-29:
Currently calling `terminal = Terminal(terminal_area, font)` only works of the font passed is `terminalio.FONT`.

Suggest allowing fixed with .bdf fonts to also work, for example:

font = bitmap_font.load_font(font_file)
terminal = Terminal(terminal_area, font)

This would allow terminals for emulation to use font specific to the emulation, for example IBM PC Code Page 437 fonts.

FoamyGuy, 2025-07-09:
@TheKitty can you share the font file? I could try to take a look into this and see if there is a simple fix. 

My gut instinct looking at the way it displays is there is some issue with the height measurement, perhaps there are ascender/descender values that aren't being treated properly.

TheKitty, 2025-07-09:
Font sent offline

FoamyGuy, 2025-07-11:
@TheKitty adding these lines just before the main loop of the example code in your previous message will move the cursor and then write a string at the new location:
```
move_cursor = chr(27) + "[10;10H"
terminal.write(f"{move_cursor}Moved to somewhere else")
```

I am not 100% certain of the arguments, I think it's x/y starting from top left of the terminal, so this is 10 characters in from left, and 10 down from top after it moves. But I have not experimented with other values to validate.

TheKitty, 2025-07-11:
Most of my time yesterday evening was getting extended characters working.
Then I tried to write them at (50,20) and it did the horizontal but not the
vertical  motion. I tried the fullscreen code with no effect

Anne Barela
Engineer, Adafruit Industries
***@***.***

FoamyGuy, 2025-07-11:
I think it may actually be y,x instead of the other way around. `"[50;20H"` stays at the bottom of the screen because 50 is larger than the number of rows in the terminal. 

`"[20;50H"` moves the cursor to this position:

[image]