adafruit/circuitpython #10450: Setting display environment variables does not set corresponding resolution for DVI

TheKitty, 2025-06-30:
### CircuitPython version and board name

```python
Adafruit CircuitPython 10.0.0-alpha.7 on 2025-06-17; Adafruit Metro RP2350 with rp2350b
Board ID:adafruit_metro_rp2350
UID:030FA9DAE6154C9A
```

### Code/REPL

```python
I cannot seem to get the settings.toml values
CIRCUITPY_DISPLAY_WIDTH=640
CIRCUITPY_DISPLAY_HEIGHT=480
CIRCUITPY_DISPLAY_COLOR_DEPTH=2

To be seen by this code. if the mode is 320x240 before the code is run, it will not toggle the screen to 640x480.
```

### Behavior

Screen remains the same, for example 320x240

tannewt, 2025-07-11:
I think these settings will apply after a restart but not a reload. We only init the display on startup. Can you confirm it works after a restart?

TheKitty, 2025-07-11:
I can, would you be more specific as to what is a restart vs what is a
reload?

PR #10493 (2025-07-18): Add restart/reload note about env variables
Fixes #10450
commit 7ec01bbd (2025-07-18): Add restart/reload note about env variables

Fixes #10450
closed 2025-07-18 by dhalbert