adafruit/circuitpython #10451: MemoryError can occur when there is still lots of free memory

FoamyGuy, 2025-06-30:
### CircuitPython version and board name

```python
Adafruit CircuitPython 10.0.0-alpha.7-6-ge3c5548344 on 2025-06-25; Adafruit Fruit Jam with rp2350b
```

### Code/REPL

```python
import gc

from displayio import Group, TileGrid, Palette
from terminalio import Terminal, FONT

from adafruit_bitmap_font import bitmap_font
import supervisor

# un-comment and the font load succeeds
#l = [1024] * 1024

main_group = Group()
display = supervisor.runtime.display
display.root_group = main_group

print(f"free: {gc.mem_free()}")
# font = FONT
font = bitmap_font.load_font("fonts/terminal.lvfontbin")
print("font loaded")
while True:
    pass
```

### Behavior

When the code is run as-is it raises this memory error:
```
code.py output:
free: 8132080
Traceback (most recent call last):
  File "code.py", line 18, in <module>
  File "adafruit_bitmap_font/bitmap_font.py", line 71, in load_font
  File "adafruit_bitmap_font/lvfontbin.py", line 77, in __init__
MemoryError: memory allocation failed, allocating 28161 bytes
```

However if you un-comment the `l = [1024] * 1024` line to use up some more memory before attempting to load the font then it succeeds. 

### Description

It feels like there is some internal boundary with the RAM that causes this error to get raised if you happen to land too close to it (or perhaps right on it?). But if you have differently sized and shaped stuff in RAM you can avoid this boundary and thus not get Memory Error. Maybe related to the PSRAM on the device?

### Additional information

This is the font file used: https://github.com/adafruit/Fruit-Jam-OS/blob/main/fonts/en_US.lvfontbin

Though I am not sure if the specific font matters so much as what is loaded before it and how big it is, I assume any font found hit this "unlucky" boundary that causes this to occur under the right circumstances.

FoamyGuy, 2025-06-30:
The line this raises from is the `f.read()` here: https://github.com/adafruit/Adafruit_CircuitPython_Bitmap_Font/blob/main/adafruit_bitmap_font/lvfontbin.py#L77 

`f` is the file that is opened as read binary here: https://github.com/adafruit/Adafruit_CircuitPython_Bitmap_Font/blob/main/adafruit_bitmap_font/bitmap_font.py#L51

tannewt, 2025-07-10:
Some internal allocations require that they occur in internal memory instead of PSRAM. I bet that's what is happening here.

How big is section_size?

tannewt, 2025-07-11:
That seems like where it is coming from. We should check to see if the file read is allocating to internal memory only.

FoamyGuy, 2025-07-11:
There are many calls to `f_read()`, and a handful to `allocate_memory()` inside of the font parsing code here: https://github.com/adafruit/circuitpython/blob/main/shared-module/lvfontio/OnDiskFont.c

I'm not sure how to tell if they are allocating to internal or not though.

tannewt, 2025-07-29:
Ok, I've figured out the issue. It happens when growing the heap with an allocation that is larger than the existing heaps total size. (This is why it works if you do the smaller 7936 allocation first. When allocating for 7937 it is growing based on the total heap size, not the in-progress allocation alone.)

We computed the size necessary for the in-progress allocation slightly wrong which resulted in expanding the heap too little, thus causing the memory error. We computed the number of blocks needed for the new allocation but didn't account for [the ATB rounding down](https://github.com/adafruit/circuitpython/blob/3f95c01335908917d513de5bacfea3cc425389ec/py/gc.c#L200). Each ATB byte stores metadata for 4 blocks. So if we need 17 we need to compute the total size for 20. Otherwise, the new heap will only have 16 and be too small.

PR #10525 (2025-07-30): Tweak computation of new heap area size
It now computes slightly more bytes to account for ATB size using truncated division. In other words, any remaining block space that doesn't fill an ATB byte won't be used. So, we round up our next area size to use an exact number of ATB bytes.

Fixes #10451
commit af672614 (2025-07-30): Tweak computation of new heap area size

It now computes slightly more bytes to account for ATB size using
truncated division. In other words, any remaining block space that
doesn't fill an ATB byte won't be used. So, we round up our next
area size to use an exact number of ATB bytes.

Fixes #10451
closed 2025-07-30 by dhalbert