adafruit/circuitpython #10455: PicoW [Errno 113] ECONNABORTED <class 'OSError'>

gsexton, 2025-07-03:
### CircuitPython version and board name

```python
Adafruit Circuit Python 9.2.8 dated 07/02/2025 on PicoW
```

### Code/REPL

```python
I have a wireshark dump I can upload to a private location if you would like. I'm pretty sure the issue is in the network stack, and not the code.
```

### Behavior

[07/02/2025 21:18:22] ERROR InfluxWriter._send_request(body=tempmon,sensor=otext temp=68.30) Exception: [Errno 113] ECONNABORTED <class 'OSError'>

### Description

So I have a device that I'm using. It has a LCD display, and a TMP102 thermometer.

It continuously polls the thermometer and once per minute makes an HTTPS POST request to place the data on a server.

I frequently get:
```
[Errno 113] ECONNABORTED <class 'OSError'>
[Errno 113] ECONNABORTED <class 'OSError'> socket_resolve_host() returned -2
[Errno 104] ECONNRESET <class 'OSError'>
```
For example, 20:12:39 failed, 20:13,and 20:14 succeeded, and 20:15:40 failed.

This was happening with 9.2.6 as well. Also, the issue happens in two different environments with quite different AP's, networks, etc.

I don't have the error message, but at one point I had a crash message where the kernel itself crashed, and the pico was locked up.

I have another one running an ancient micropython version, and it just works. Is there any hope for this ever working?

### Additional information

When it starts failing, I see a lot of SYN, SYN/ACK retransmits between the two hosts. I can provide you with the tcpdump capture file if you would like. It has both successful, and failing connection cycles.

gsexton, 2025-07-04:
You can grab the sanitized pcap file from:

https://www.mhsoftware.com/~gsexton/sanitized.pcap

gsexton, 2025-07-04:
My apologies. I went through and updated all of the libraries, try 10 alpha 7, and it seems to be working as expected now.

closed 2025-07-04 by gsexton

gsexton, 2025-08-01:
I'm re-opening this because it's still present. I'm using 10.0.0 beta 2. I've updated all python code that's running from the source repos, and re-compiled using the current mpy-cross command line utility. I routinely get:

```
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```
My program is reasonably large, and this error happens after 8-10 hours of operation. If you can give me an idea on how to localize the location more, I'll try that.

Addtionally, the connection issues persist:

```
[07/31/2025 04:20:58] ERROR InfluxWriter._send_request(body=tempmon,sensor=otext temp=69.83) Exception: [Errno 113] ECONNABORTED <class 'OSError'>
[07/31/2025 04:23:00] ERROR InfluxWriter._send_request(body=tempmon,sensor=otext temp=69.89) Exception: [Errno 113] ECONNABORTED <class 'OSError'>
[07/31/2025 04:25:00] ERROR InfluxWriter._send_request(body=tempmon,sensor=otext temp=69.80) Exception: [Errno 113] ECONNABORTED <class 'OSError'>
```
What's really sort of interesting is that every other attempt fails. Looking above, you can see 4:21:00, and 4:24 aren't represented. This is consistently seen.

It would be really nice if this could be made to work.

gsexton, 2025-08-05:
It's a local server. I wrote a test case that makes 10 HTTPS POST requests per second, and right now it's ticking past 9,000 with no errors.

The application has two other components. It has a ST7789 display that it writes data to, and a TMP102 I2C sensor that it reads once per second.

@dhalbert Is there any way that I can debug where in my main code set the hard fault is happening? I have a Pico debugger if that's useful.

```
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
```

dhalbert, 2025-08-05:
You could set a breakpoint on `reset_into_safe_mode()`, and then look at the stack backtrace. It would be good to compile in debug mode so you can get more info. But sometimes that eliminates a bug because the code or timing is different.

I would say use your original program, but just speed up the POSTing to be every time you read the temperature sensor. If you can then get it to fail, then comment out the display code and see what happens, and if that doesn't cause the problem to go away, then uncomment the display code and comment out the TMP102 reading. The idea is to eliminate causes if possible.

gsexton, 2025-08-06:
The content of post_test.php is:

```php
<?php

ob_implicit_flush(1);
function TestPost() {
    $wrValue = array("SomeField" => "SomeValue","OtherField" => 42);
    echo json_encode($wrValue);
    flush();
    
}
TestPost();
?>
```

gsexton, 2025-08-07:
I took the code, and commented out the SSL bits and changed the URL to non-SSL. The TCP/IP stack is just rubbish, and this proves it. Here's the output:

```
192.168.168.126 | code.py | 10.0.0-beta.2Request counter=10wrValue);
Request counter=20
Request counter=30
Request counter=40
Request counter=50
Request counter=60
Request counter=70
Request counter=80
Request counter=90
Request counter=100
Request counter=110
Request counter=120
Request counter=130
Request counter=140
Request counter=150
Request counter=160
Request counter=170
Request counter=180
Request counter=190
Request counter=200
Request counter=210
                                 Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```

gsexton, 2025-08-07:
I stripped out the SSL connection but it still crashes with the simplified code. So either the TCP stack is unstable, or just the whole of circuitpython is unstable.

```
Request counter=47980
Request counter=47990
                                     Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```

dhalbert, 2025-08-07:
> So either the TCP stack is unstable, or just the whole of circuitpython is unstable.

The structure of the code is that the `wifi` API's use https://github.com/georgerobotics/cyw43-driver underneath, to talk to CYW43 radio module. That uses lwip. Since you ruled out TLS, we don't have to look at mbedTLS, which is also used.

So it looks like it takes a few tens of thousands of calls on average to hit the bug. Is there a high variance in the number of calls? Did you add the `gc.collect()` call because you were getting memory errors, or is it just for instrumentation.

Could you post the HTTP-only version as well? We'll try to reproduce starting with that.

gsexton, 2025-08-08:
I spent a lot of time troubleshooting memory problems with SSL. On a pico w, I could make SSL work with one certificate in my application, but if I tried to use two, I ran out of memory immediately. My "fix" was eventually to have my site grab data from the 2nd site and serve it. The GC stuff is left over from that.

```python
import adafruit_requests
import json
import socketpool
import ssl
import time
import wifi
import gc

ssid="myssid"
psk="mypass"
while True:
    try:
        res = wifi.radio.connect(ssid, psk)
        print("Connect result=", res)
        if wifi.radio.connected:
            print("Connected to wifi!")
            break
        print("Trying connection again.")
        time.sleep(1)
    except Exception as e:
        print(e)
        time.sleep(1)

pool = socketpool.SocketPool(wifi.radio)
ctx = ssl.create_default_context()
requests = adafruit_requests.Session(pool, ctx)

print(f"gc.mem_alloc()={gc.mem_alloc()} gc.mem_free() = {gc.mem_free()}")

gc.collect()
print(f"Initial Post Cert Load: gc.mem_alloc()={gc.mem_alloc()} gc.mem_free() = {gc.mem_free()}")
data={"SomeValue": "SomeString", "FloatValue": 0.2, "IntValue": 42, "Slice": ["a","b","c"]}
counter = 0
while True:
    try:
        response = requests.post("http://somehost.somedomain.com/post_test.php", json=data, timeout=10)
        rdata = response.json()
        response.close()
        del response
        time.sleep(0.1)
        counter+=1
        if counter%10==0:
            print(f"Request {counter=}")
    except Exception as e:
        print(f"Exception: {str(e)}")
        time.sleep(1)
```

dhalbert, 2025-08-08:
@gsexton Just to let you know - we appreciate this testing, and it's on our list to look at this. If you are blocked by these problems, switching to a Pico 2 W would help for now, but there should be fixes on Pico W as well. There are limitations based on the amount of RAM on RP2040, but we could do better.

dhalbert, 2025-09-03:
I have been trying to reproduce this on a Pico W with a very simple local webserver, running in Python, and an even simpler version of your HTTP test program in https://github.com/adafruit/circuitpython/issues/10455#issuecomment-3166290627, doing 10 requests a second. I have seen cases where the `wifi` module code starts raising EINPROGRESS errors, but mostly it is works fine.

_However_ (and this is a big "however"), I am consistently seeing a crash into safe mode when I attempt to edit `code.py` while the program is running, or write another larger file. This matches your more recent testing. So there is some kind of core bug here which I will investigate.

dhalbert, 2025-09-05:
I am able to reproduce this fairly easily with repeated HTTP fetches (doesn't have to be POST). I think this may be due to PR #10027, which switched to dynamic storage allocation in LWIP. I was able to get a stack backtrace after a failure that reported a `tlsf` assertion failure and I see that LWIP is doing storage allocate/free during an interrupt context. The `port_malloc()` and `port_free()` calls it uses are not protected in a critical section, so there may well be some overlapping storage operations.

#10027 was added between 9.2.4 and 9.2.5. Sure enough, 9.2.4 has been running happily though slowly for thousands of operations, but I was able to get a pretty quick crash in 9.2.5.

There were a lot of changes in #10027, so it may be something else, but it's narrowed down now.

PR #10615 (2025-09-06): RP2xxx: put LWIP storage ops in a critical section
- Fixes #10455 

#10027 (added in 9.2.5) considerably improved network performance on Pi Pico by using dynamic storage allocation in LWIP. However, it turns out LWIP can do heap operations during interrupts. Thus the operations need to be guarded in critical sections. Otherwise there are storage-related crashes at random intervals, and I saw storage assertion violations in gdb when a crash happened.

Tested on a Pico W, with a simple test program that did an HTTP fetch from a local server 10 times a second.

On a Pico 2 W, I was unable to caus [... cut]