adafruit/circuitpython #10456: Reducing freeze time for gc.collect()

jbrelwof, 2025-07-05:
**gc.collect()** freezes things for a sizeable chunk of time.  How can that be improved?
 

- faster measurement
  - optimization without good measurements is... challenging
  - **gc.mem_alloc()** and gc.mem_free() are terribly slow - they use  **gc_info()** (in C) to add up stats for each block in each **mp_state_mem_area_t**
    - adding a singleton to track the same info would add a little overhead in time for allocating and collecting instances, but would dramatically improve the speed of  **gc.mem_alloc()** and **gc.mem_free()**
    - half of the hooks (allocation side) to do this appear to already be in place for the [memorymonitor](https://docs.circuitpython.org/en/stable/shared-bindings/memorymonitor/) module,  adding some tracking for frees during collection shouldn't be too difficult
- control when collection happens
  - **gc.disable()** and manually call **gc.collect()**
  - at least manages when the pause will be, but not how long it will be
  - would benefit greatly from faster measurement to determine when it needs to be done
- control how long collection takes
  - this is probably the most important improvement for getting things under control
    -  but might be less effective without manually initiating **gc.collect()**
    - requires the ability to do partial/incremental collection.  It might not be possible without significant rework in the micropython GC code
      - but should be far less work than supporting reference counting
      - and could potentially remain a compile time configuration
- minimize how often collection is required by avoiding new allocations
  - combined with manual control of collection timing, this can keep things as predictable as possible (although faster measurement certainly helps)
  - but at least for my use case, it's not practical to eliminate this entirely since using `*args, **kwds` in parameters or invocations usually creates temporaries that will have to be collected

dhalbert, 2025-07-05:
In the latest alphas of 10.0.0, we've eliminated scanning for objets inside objects that contain no pointers (bytearrays, etc.). This speeds up many collections substantially. It's worth trying to see how it affects your particular program. See #10264.

jbrelwof, 2025-07-05:
> You can also use [`gc.threshold()`](https://docs.circuitpython.org/en/latest/docs/library/gc.html#module-gc) to invoke automatic gc earlier, when it will take less time.

I was going to give it a try, but unfortunately [`gc.threshold()`](https://docs.circuitpython.org/en/latest/docs/library/gc.html#module-gc) doesn't appear to be available in the 10.0 alpha build I'm using.

`dir(gc)=['__class__', '__name__', '__dict__', 'collect', 'disable', 'enable', 'isenabled', 'mem_alloc', 'mem_free']`

jbrelwof, 2025-07-11:
Turns out that one of the main culprits in my use case was frequent calls to **time.monotonic_ns()**, which returns a long long int (and therefore causes an allocation which must be collected, as opposed to regular CircuitPython int/float values).   When I swapped those out for calls to **time.monotonic()** (which returns a float _without_ triggering an allocation) my calls to **gc.collect()** became significantly faster (from an average of ~350 down to ~70 milliseconds).

If **gc.collect()** stays under a tenth of a second, that might be "good enough" for my current project, at least for a while.   I still would like to see (and might try implementing) a much more efficient implementation for **gc.mem_alloc()** and **gc.mem_free()** - currently, each call averages around 20 milliseconds.  That's tolerable for running an occasional "is it time for another **gc.collect()** test, but starts adding up quickly when trying to profile nested memory use to identify which part of the code is leaving trash on the floor.

tannewt, 2025-07-11:
When I was implementing the collect optimization I was using C-level pin high and low toggles to use my logic analyzer to measure collect performance. It is about as low overhead as you can get I think.

You may be able help things by finding temporary allocations that can be manually freed.

What microcontroller are you on?

jbrelwof, 2025-07-12:
> I think we disable `gc.threshold()` because it isn't in CPython's `gc`: https://docs.python.org/3/library/gc.html

It's still in the [CircuitPython gc docs](https://docs.circuitpython.org/en/latest/docs/library/gc.html#gc.threshold) though.  It's not a big deal for me at the moment because I'm running under `gc.disable()` and manually calling `gc.collect()`, but I could see it being useful.

jbrelwof, 2025-07-13:
> When I was implementing the collect optimization I was using C-level pin high and low toggles to use my logic analyzer to measure collect performance. It is about as low overhead as you can get I think.

I don't recall the exact numbers, but I saw some major improvements when I switched from 9.x to the "latest build" 10.0.alpha with the new optimizations.

I'm measuring `gc.collect()` time with before/after calls to `time.monotonic()`.  That adds a little overhead, but shouldn't be significant enough to matter when measuring tens of milliseconds.  The actual timings I'm getting at the moment show `gc.collect()` have increased to around 220 milliseconds to free about 1.2M of ram, leaving around 305K used.  It had dropped to ~70ms, but it keeps fluctuating add more load to the "main loop".  I'm not sure whether that's unusually long for that much collection,  but it's much better than it was before I stopped using `time.monotonic_ns()`

 
> You may be able help things by finding temporary allocations that can be manually freed.

What do you mean by 'manually freed'?   I assumed `del somePythonObject`  couldn't really do much in a GC (vs reference counted) based runtime - is there something I'm missing?  I have put in quite a bit of effort into reworking things to reduce / eliminate temporary allocations in the first place, which has helped.   But some things ( *arg / **kwds overhead ) can't always be avoided, and all the extra effort becomes much less significant as soon as I start pushing and JSON updates out on websockets...

> What microcontroller are you on?

primarily ESP32 variants (S2, S3, C3, C6, ...) typically with at least 2MB ram