adafruit/circuitpython #10461: Initializing ESP_SPIcontrol after DAC results in audio not playing

FoamyGuy, 2025-07-07:
When using both DAC and ESP32SPI it seems that the order the objects are initialized causes the DAC to work or not. If `ESP_SPIControl` is initialized before the DAC then audio play back works successfully. But if the DAC is initialized before `ESP_SPIControl` then audio playback does not work. `audio.playing` property is true and the while loop finishes after the length of the audio file, but no audio comes out of the speaker. 

```
Adafruit CircuitPython 10.0.0-alpha.7-7-g363be5fe80 on 2025-07-01; Adafruit Fruit Jam with rp2350b
```

This reproducer script illustrates the issue. Swapping between the two commented ESP init lines causes the audio playback to work or not work consistently for me on Fruit Jam revC hardware. 
```

from audiocore import WaveFile
import time
import gc
import board
import busio
from digitalio import DigitalInOut
from adafruit_esp32spi import adafruit_esp32spi
from os import getenv
import adafruit_tlv320
import audiobusio

ssid = getenv("CIRCUITPY_WIFI_SSID")
password = getenv("CIRCUITPY_WIFI_PASSWORD")
esp32_cs = DigitalInOut(board.ESP_CS)
esp32_ready = DigitalInOut(board.ESP_BUSY)
esp32_reset = DigitalInOut(board.ESP_RESET)
spi = busio.SPI(board.SCK, board.MOSI, board.MISO)

# Init ESP_SPIControl here and the audio beep plays successfully
esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)

i2c = board.I2C()
dac = adafruit_tlv320.TLV320DAC3100(i2c)
dac.headphone_output = True
dac.configure_clocks(sample_rate=16000, bit_depth=16)
audio = audiobusio.I2SOut(board.I2S_BCLK, board.I2S_MCLK, board.I2S_DIN)
wave_file = open("/beep.wav", "rb")
beep_wave = WaveFile(wave_file)

# Init ESP_SPIControl here and the audio beep does not play
# esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)

time.sleep(2)

audio.play(beep_wave)
while audio.playing:
    pass

print("Audio playback complete")

```
This audio file is used: [beep.wav.zip](https://github.com/user-attachments/files/21109474/beep.wav.zip)

dhalbert, 2025-07-07:
For debugging help, if you initialize I2C first, does it work or not?
```py
# moved to before esp32spi init
i2c = board.I2C()
dac = adafruit_tlv320.TLV320DAC3100(i2c)

esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)
```

FoamyGuy, 2025-07-07:
With the code like this is also does not play the audio:
```

from audiocore import WaveFile
import time
import gc
import board
import busio
from digitalio import DigitalInOut
from adafruit_esp32spi import adafruit_esp32spi
from os import getenv
import adafruit_tlv320
import audiobusio

ssid = getenv("CIRCUITPY_WIFI_SSID")
password = getenv("CIRCUITPY_WIFI_PASSWORD")
esp32_cs = DigitalInOut(board.ESP_CS)
esp32_ready = DigitalInOut(board.ESP_BUSY)
esp32_reset = DigitalInOut(board.ESP_RESET)
spi = busio.SPI(board.SCK, board.MOSI, board.MISO)

i2c = board.I2C()
dac = adafruit_tlv320.TLV320DAC3100(i2c)
dac.headphone_output = True
dac.configure_clocks(sample_rate=16000, bit_depth=16)

# Init ESP_SPIControl here and the audio beep plays successfully
esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)

audio = audiobusio.I2SOut(board.I2S_BCLK, board.I2S_MCLK, board.I2S_DIN)
wave_file = open("/beep.wav", "rb")
beep_wave = WaveFile(wave_file)

# Init ESP_SPIControl here and the audio beep does not play
# esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)

time.sleep(2)

audio.play(beep_wave)
while audio.playing:
    pass

print("Audio playback complete")

```

I will try moving each line one at a time and see which if any done before ESPSPI_Control will allow audio to play.

FoamyGuy, 2025-07-07:
This order of initializing lets audio play successfully:
```
i2c = board.I2C()
dac = adafruit_tlv320.TLV320DAC3100(i2c)

# Init ESP_SPIControl here and the audio beep plays successfully
esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)

dac.headphone_output = True
dac.configure_clocks(sample_rate=16000, bit_depth=16)

audio = audiobusio.I2SOut(board.I2S_BCLK, board.I2S_MCLK, board.I2S_DIN)
wave_file = open("/beep.wav", "rb")
beep_wave = WaveFile(wave_file)
# ...
```

From here, moving `dac.headphone_output = True` to be above the ESP_SPIControl init makes the audio no longer work.

FoamyGuy, 2025-07-07:
Do the IRQ pins for I2S or ESPSPI get used "automatically" internally somehow?

I noticed that both I2S and ESP have IRQ assigned to the same pin:

dhalbert, 2025-07-08:
@FoamyGuy good sleuthing! I remember now seeing they were shared when I was working on `pins.c` for the latest version. There were no spare pins and these ended up shared. We'll need to look at the library code and the `audiobusio` code to see what's going on here.

@ladyada FYI

dhalbert, 2025-09-15:
The reason for this is that the TLV320DAC3100 and the ESP32-C6 shared a reset line, `board.PERIPH_RESET` (aka `board.ESP_RESET`) on the Fruit Jam. To set up the C6 requires pulling its reset line low, which will reset the state of the TLV320.

The TLV320 does not need to be hard-reset in general. You can reset it over I2C with a RESET command, and there's nothing in the library that deals with the reset pin.

I will add a warning to the Fruit Jam guide to init the ESP32-C6 first, and that any re-resets will require setting up the TLV320 again.