adafruit/circuitpython #10491: Change appropriate m_new() to m_malloc_without_collect()

dhalbert, 2025-07-16:
This is a follow-on to #10264.

There are a number of places where `m_new()` is called to create byte or string buffers, etc., which contain no collectible objects. An example is `py/vstr.c`. Change these to m_malloc_`without_collect()`

mikeysklar, 2026-11-02:
@dhalbert can we close this? It seems resolved by #11708.

dhalbert, 2026-11-02:
@mikeysklar Did you you find any other `m_new()`s that could be changed to `m_malloc_without_collect()`? I did not do any kind of inventory when I filed the issue.

mikeysklar, 2026-11-02:
I did an inventory to determine remaining `m_new()` calls we can swap out.

The biggest leftovers are audio buffers: 

- Freeverb comb and allpass (about 25 KB mono)
- the MultiTapDelay delay line (up to 128 KB)
- Flanger
- AudioFileWriter 32 KB ring

Smaller ones are:

- zlib dictionary and output
- lvfontio glyph cache
- mpz digits

Should I add these to this PR?

dhalbert, 2026-11-02:
I merged #11708 already. A new PR with all of these would be great - thanks.

mikeysklar, 2026-11-02:
Larger s/m_new/m_malloc.../g in progress ETA 2h.

closed 2026-11-02 by dhalbert