adafruit/circuitpython #10511: rp2pio.StateMachine valid frequency has changed

FoamyGuy, 2025-07-25:
The code here: https://github.com/adafruit/Adafruit_CircuitPython_PIOASM/blob/main/examples/pioasm_hello.py#L30 which is embedded in this guide page: https://learn.adafruit.com/intro-to-rp2040-pio-with-circuitpython/using-pio-to-turn-an-led-on-and-off

Creates a StateMachine with frequency `2000`. When I run this code on a Pico w/ 10.0.0-beta.0 it raises this error: `ValueError: frequency out of range`

In testing a few values it seems that `4000` is the minimum accepted value. 

I can update the example code to get it back to working if this change was intentional. But I'm not sure if this was intentional or not, so am opening the issue here to get more input on that. 

I found information online (https://www.bisonacademy.com/ECE476/Lectures/29%20PIO%20State%20Machines.pdf) indicating the valid range is 2k - 125k which seems to match the lower end value of `2000` originally used by the basic example

[image]

I also noticed that if I use `frequency=0` the code prints `real frequency 252000000` which is basically 2x the 125k listed as the top of end of the valid range. So perhaps there is some multiplier taking effect on the valid range? Maybe this is the tied to RPi specifying the chip was usable at higher CPU frequencies earlier this year?

In any case, if this change is intentional we can close this and I'll update the learn code use `4000` so that it runs.

jepler, 2025-07-25:
In the PIO hardware, the clock rate is created by applying a divisor to the CPU clock ([microcontroller.Processor.frequency](https://docs.circuitpython.org/en/latest/shared-bindings/microcontroller/index.html#microcontroller.Processor.frequency)). So there is not a fixed lower bound, it depends on the system clock. Also, the lower bound is not exactly 2kHz or 4kHz.

The maximum divisor is 0x1_0000 (65536). For the original non-overclocked RP2040 at the default frequency of 125MHz, this meant speeds as low as 120_000_000Hz/0x1_0000 ~= 1831Hz were available. For RP2350 operating at the default frequency of 150MHz, the lowest clock is ~2289Hz instead. For overclocked devices, the lowest PIO frequency becomes even higher.

Page 947 of https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf documents the underlying hardware register, SM0_CLKDIV.

FoamyGuy, 2025-07-25:
Okay, thank you for confirming and the pointer to docs. I'll close this and submit a PR to update the example code. 

Thanks also for the guide and it's many examples. I'm just starting to look closer into PIO and it's been quite helpful.