adafruit/circuitpython #10522: does  last release of circuitpython does  not support  ftp service

Geromino, 2025-07-29:
A CircuitPython 9.1.0-beta.3-6-gc2443f993f
 B CircuitPython 9.2.4-154-gbd572f86f9
when  i compile   A  create uf2   ftp service appears via nrfconnect mobile app 
when i compile  B  create uf2   ftp  service  does not appears  nrf connect mobile app 

1. does  circuitpython  stop support  ftp service  ?  
2. does  it possible  add  to B   then prepare again uf2 
3. or i need to stay with A ?

dhalbert, 2025-07-29:
As of #9344 (included in 9.1.0-beta.4), we no longer always start the BLE workflow, including the file transfer service you mention. Read that PR and see https://docs.circuitpython.org/en/latest/docs/workflows.html#ble for information about how to enable the BLE workflow when booting.

Geromino, 2025-07-30:
hi  ,  done   following but nothing  was change  am  still not see ftp service  in nrfconnect  mobile app 
i  rebuild my uf2  with changes in file mpconfigboard.mk
USB_VID = 0x239A
USB_PID = 0x80DA
USB_PRODUCT = "PCA10056"
USB_MANUFACTURER = "Nordic Semiconductor"

MCU_CHIP = nrf52840
UF2_NAME = "SENSERA_Device"
CIRCUITPY_BUILD_EXTENSIONS = bin,uf2

QSPI_FLASH_FILESYSTEM = 1
EXTERNAL_FLASH_DEVICES = "GD25Q16C, W25Q16JVxQ"
CIRCUITPY_BLEIO = 1
CIRCUITPY_BLE_WORKFLOW = 1
CIRCUITPY_BLE_FILE_SERVICE = 1

also   i  add  file in circuitpy  drive  settings.toml  CIRCUITPY_BLE_ENABLED = 1  
in link  that you send  i didnt see these steps  i found reference in  workflow  circuitpython   please advise

dhalbert, 2026-02-07:
Closing for now.