adafruit/circuitpython #10524: "SD_SCK in use" when trying to use SD SPI on Metro RP2350

todbot, 2025-07-29:
### CircuitPython version and board name

```python
Adafruit CircuitPython 10.0.0-beta.0 on 2025-07-15; Adafruit Metro RP2350 with rp2350b
```

### Code/REPL

```python
import time, board, busio, sdcardio
try_adafruit_sdcard = True
mount_point = "/sd"

# pin definitions
sd_sck   = board.SD_SCK   # board.GP34   # "CLK"
sd_mosi  = board.SD_MOSI  # board.GP35   # "SI"
sd_miso  = board.SD_MISO  # board.GP36   # "SO"
sd_cs    = board.SD_CS    # board.GP39   # "CS"

# sd card setup
if try_adafruit_sdcard:
    print("trying adafruit_sdcard")
    import adafruit_sdcard
    sd_spi = busio.SPI(clock=sd_sck, MOSI=sd_mosi, MISO=sd_miso)
    sdcard = adafruit_sdcard.SDCard(sd_spi)
else:
    print("trying sdcardio")
    sd_spi = busio.SPI(clock=sd_sck, MOSI=sd_mosi, MISO=sd_miso)
    sdcard = sdcardio.SDCard(sd_spi, sd_cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, mount_point)
while True:
    for filename in os.listdir(mount_point):
        print("filename:",filename)
    time.sleep(0.5)
```

### Behavior

```
code.py output:
trying adafruit_sdcard
Traceback (most recent call last):
  File "code.py", line 15, in <module>
ValueError: SD_SCK in use
```

### Description

There seems to be no way to release the SD SPI pins being used.
Trying to do a minimal version of the [SD Card Read Test in the Learn Guide](https://learn.adafruit.com/adafruit-metro-rp2350/sd-card#sd-card-read-test-3153413) results in "SD_SCK in use" error.

dhalbert, 2025-07-29:
Yes, we have to document this, and yes, it uses `sdcardio`. You can do `storage.umount("/sd")` to unmount, and then mount again however you wish.

@tannewt can say more as well.

todbot, 2025-07-29:
If I wanted to unconfigure all the auto-configured stuff on the Metro RP2350, so far I know to do:
```py
displayio.release_displays()
storage.umount("/sd")
usb_host.???
```
Is there a method to unconfigure the PIO-USB usb host stuff?