adafruit/circuitpython #10527: Analog input keys for the keypad module

rivimey, 2025-07-29:
I have built a board that includes 4 keyswitches connected to an analog input on an rp2040 mcu. I have a function which reads the analog converter and decodes the result into 4 different keyup / keydown actions.

I would like to simplify the rest of my code by using an event abstration rather than if-blocks, and saw the keypad module includes "Event" and EventQueue, which sounded great -- until I saw that it was not instantiable in normal code.

So... I guess I could reimplement EventQueue in python, though I worry that would cause memory allocation thrashing. However the obvious thing would be to add in an AnalogKeys variant of the keypad module. My existing code is already "programmed" using an array of min-max values and a bitmask of keys, (which it is indeed annoying to prepare, but only needs doing once.)

What I don't know is how to go about making an AnalogKeys class a reality, even just for me. Can you suggest a direction to go in?

todbot, 2025-07-29:
I'd say try creating a library that borrows concepts from [`adafruit_debouncer`](https://github.com/adafruit/Adafruit_CircuitPython_Debouncer) and provides you with press & release events. I think `adafruit_debouncer` would *almost* work (as it can take an arbitary lambda), but it would be inefficient. Another approach would be to find an existing Arduino library (maybe [ADCButtons](https://github.com/An7orAhmed/ADCButtons)?) that matches your circuit and translate that to a CircuitPython library. 

If you'd like help creating a CircutPython library, I'd be happy to help. We could continue on a thread in the #help-with-circuitpython Discord, as this repo is more for internal C-based development of CircuitPython.

rivimey, 2025-07-30:
To be clear, I already have code that works, albeit not in an "event" style, so I don't need another solution for that.

What I was asking about is whether
(a) it is possible for me to add an AnalogKeys class to circuitpython core, or preferably
(b) how would the existing code be made usable from python context, so I can create my new class in python?

I say preferably (b) because it would enable lots of other use cases and simplify tinkering, but (a) works for me too.

dhalbert, 2025-07-30:
Could you say more exactly what you are doing with the inputs? You're reading a variable analog voltage and interpreting it? What exactly is the mapping? To add an `AnalogKey` capability, it would be something that is pretty widely applicable.

The main virtue of `keypad` is that it does background scanning so you don't have to. Or, are you willing to do the scanning yourself, but you'd like to add events to an `EventQueue`? We've talked about making it possible to create your own `EventQueue` objects, but haven't done that yet.

Neradoc, 2025-07-30:
Do you need background scanning ? If so, you would have to try and make a class for the builtin keypad library in C.
If not, and you can scan "manually" often enough in your loop, you can implement it in python.

I wrote this, I believe it worked at the time, but I haven't used it since actually.
https://github.com/Neradoc/Analog_Button_Keypad/blob/main/analog_button.py
> Helper class for buttons using a resistor ladder and analog output. Follows the keypad module API.

Also this to combine multiple event queues:
https://github.com/Neradoc/Circuitpython_Multi_Keypad

(Requires manual scanning)

todbot, 2025-07-30:
Ha, Neradoc, wish I'd seen your analog_button library before I did a quick swipe at one too yesterday: [analog_buttons.py](https://gist.github.com/todbot/fe9bfb4eebbd52d4d2e5f000b01d46f7) It also implements an Event queue but I'm not sure it's all that necessary as it's not like you can detect multiple button presses

rivimey, 2025-07-31:
> The main virtue of `keypad` is that it does background scanning so you don't have to. Or, are you willing to do the scanning yourself, but you'd like to add events to an `EventQueue`? We've talked about making it possible to create your own `EventQueue` objects, but haven't done that yet.

Basically I am asking how to go about making an update to the keypad module/class group that supports AnalogKeys, where a key is defined in the fairly generic way shown in my code snippet above (i.e. a table of min-max adc readings and a bitmask result).

It would be great if you could make EventQueue more available too.