adafruit/circuitpython #10531: Add a setting in settings.toml to control terminal display in connected displays

rbrian, 2025-07-30:
On those boards that have a built-in display (and even those that don't, when a display has been connected but then the board restarts) there are several lines of text shown in the terminal before there's an opportunity to turn off the display in boot.py. For some projects, having this display eliminated would result in a much more usable project. 

My current project is a kids' toy, based on an ESP32S3-TFT feather. Having that text displayed wouldn't be a good thing in this situation 

A suggestion was made in discord to have a setting in settings.toml to control whether or not the default terminal display would be shown. This would be a very good & clean solution as far as I can tell. 

Make it so. :)

dhalbert, 2025-07-31:
This is an example of wanting to use a boolean value in `settings.toml`, which is currently not supported, but would be helpful. #9113 has some related discussion about this.

rbrian, 2025-07-31:
i guess just requiring ```HIDE_DISPLAY="YES"`` or "TRUE" would be better. Yes, an actual boolean would be better, but it didn't sound from that other issue that booleans would be working any time soon?

dhalbert, 2025-07-31:
I am interested in adding booleans fairly soon, but it's not going to be immediate: we are busy getting the last issues closed for 10.0.0 final. For right now, your best bet would be to make a custom build to turn off the display.