adafruit/circuitpython #10541: Blank Screen T Deck Plus

Zoxs97, 2025-08-02:
I am using the example from the T Deck repository but it doesnt seem to work for T Deck Plus. Is there any where i can find the correct settings and libraries fpr the T Deck Plus or even better examples? I can only manage to make the screen backlight light up but nothing display on the screen

RetiredWizard, 2025-08-02:
If you're talking about the Lilygo repository those examples are Arduino, not CircuitPython. You'll probably want to use one of the Lilygo support forums for help with those.

If you want to use CircuitPython the first thing you'll need to do is flash the device with CircuitPython. That can be done from the circuitpython.org website. Click on Downloads, search for T-Deck and then use of the "Open Installer" buttons. 

If you have any questions on using the installer or running CircuitPython code your best bet would be to go to the #help-with-circuitpython discord channel at http://adafru.it/discord

Zoxs97, 2025-08-02:
Thank you i'll join their discord server!

closed 2025-08-02 by Zoxs97