adafruit/circuitpython #10544: D10 on header but not defined in board.D10

b-blake, 2025-08-04:
Per Carter, the 10 label on the 32 pin header refers to port D10, GPIO unknown att.
D10 is not defined as board.D10.  I would guess GPIO10 to conform to D6, D7, D8, & D9.

samblenny, 2025-08-05:
Are you sure about this? It looks to me like [ports/raspberrypi/boards/adafruit_fruit_jam/pins.c](https://github.com/adafruit/circuitpython/blob/f86bf4c0853ae293e3caac9bdcd49785a42ac69e/ports/raspberrypi/boards/adafruit_fruit_jam/pins.c#L27) defines D10 as GPIO10. The Fruit Jam [schematic](https://github.com/adafruit/Adafruit-Fruit-Jam-PCB) shows that D10 on the 2x16 header socket goes to GPIO10.

When I checked just now in the REPL (CircuitPython 10.0.0-beta.2), I got this:
```
Adafruit CircuitPython 10.0.0-beta.2 on 2025-07-30; Adafruit Fruit Jam with rp2350b
>>> import board
>>> hasattr(board, 'D10')
True
```

closed 2025-08-06 by tannewt

b-blake, 2025-08-07:
@dhalbert , @tannewt 

Big OOPS on my part.  D10 is buried in this line of tags and I missed it.
**'CKN', 'CKP', 'D0N', 'D0P', 'D10', 'D1N', 'D1P', 'D2N', 'D2P'** 
I apologize.  I'll do better next time.