adafruit/circuitpython #10548: PIN in use

daniel-alsen, 2025-08-06:
### CircuitPython version and board name

```python
Adafruit CircuitPython 10.0.0-beta.2 on 2025-07-31; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM
```

### Code/REPL

```python
import digitalio
import board

try:
  i2c_power = digitalio.DigitalInOut(board.I2C_POWER)
  i2c_power.direction = digitalio.Direction.OUTPUT
  i2c_power.value = False
  time.sleep(0.1)
  i2c_power.deinit()
except Exception as e:
  log(f"Failure turning off I2C power: {e}", level="ERROR")
```

### Behavior

I get the error message "I2C_POWER in use" when turning off power before going to deep sleep.

I got similar error messages for NEOPIXEL_POWER, but that seems to work now. I also get "PIN in use" error messages when setting a deep sleep pin alarm for a pin I have initialized and used in my code.

### Additional information

This uses to work in the previous 9.x version,

dhalbert, 2025-08-07:
> Is it so that you always need to deinit() a pin after usage, even if you are only using it temporarily in a local function without maintaining it in a global variable?

If you're really done with it, then you should deinit it. Internally there is storage and state to make the pin work, and there are hardware settings. So create the `DigitalInOut` object once and pass it around. It doesn't have to be literally global, but it needs to be shared.

closed 2025-08-07 by dhalbert

daniel-alsen, 2025-08-07:
Got it, thank you!