adafruit/circuitpython #10551: USB Host Improvement Quest Umbrella Issue

samblenny, 2025-08-06:
I want to make the USB host implementation more useable for game and synth projects. I'm posting this issue in the hope of getting feedback and core dev buy-in for my list of stuff I'm hoping to fix:

- [ ] **Unreliable Unplug Event Detection**: There's currently no good way to detect an unplugged USB device (some exceptions resulting from an unplug look the same as other types of timeouts). Desktop PyUSB handles this by setting `usb.core.USBError.errno` and `usb.core.USBTimeoutError.errno`.
- [ ] **Corrupted Descriptor Parsing**: I started using my own USB device descriptor parser because I would intermittently get random garbage if I used `usb.core.Device` to do descriptor parsing. I suspect possible memory corruption bugs in the `.idVendor`, `.idProduct`, `.serial_number`, `.product`, and `.manufacturer` properties of `usb.core.Device`.
- [ ] **Dropped/Stuck MIDI Notes**: When reading MIDI messages from USB MIDI controllers, sometimes note on or note off messages mysteriously go missing. (also happens for other message types, but note on/off is the most problematic)
- [ ] **Polling Uses Lots of CPU**: Anecdotally, it seems like polling for USB host device input events uses a lot of CPU time (not great when also using displayio, picodvi, syntio, etc). For example, when foamyguy ported my Pumpkin Toss game to use I2C gamepad input instead of USB gamepad input, the animation loop ended up running way faster. Maybe this be improved by using asyncio or something?

I plan to start looking at some of this stuff and hopefully submit some PRs if I can fix any of it.

Does anybody (Scott, Dan, and foamyguy in particular) have any feedback on this? Is there something else about USB host that you think needs attention? Is there a particular board I should pay attention to for making sure my proposed changes don't increase code size too much (perhaps RP2040 USB Host Feather?). Any other advice on things to pay attention to so the results are mergeable?

samblenny, 2025-08-06:
Some other issues with interesting discussion about USB host stuff:
- https://github.com/adafruit/circuitpython/issues/5423
- https://github.com/adafruit/circuitpython/issues/10494
- https://github.com/adafruit/circuitpython/issues/10243
- https://github.com/adafruit/circuitpython/issues/9930 (possible missing null pointer check in `find()`?)
- https://github.com/adafruit/circuitpython/issues/9905
- https://github.com/adafruit/circuitpython/issues/8886

tannewt, 2025-08-07:
Please make four separate issues. Responding to parts of the single post is tricky.

samblenny, 2025-08-09:
found another one: https://github.com/adafruit/circuitpython/issues/10562 (USB Host: Fast polling + write to CIRCUITPY crashes board)