adafruit/circuitpython #10555: USB Host: Polling Uses Lots of CPU

samblenny, 2025-08-07:
Currently (10.0.0-beta.2 on Fruit Jam), it seems like polling for USB host device input events with `usb.core.Device.read()` uses a lot of CPU time compared to reading GPIO pins or I2C. I know this is vague. All I have at this point is anecdotal observations that USB polling in combination with picodvi and/or synthio appears to load down the RP2350 CPU (slow display updates, dropped notes, etc). For example, last year (around 9.1.4), when foamyguy ported my [Pumpkin Toss](https://adafruit-playground.com/u/SamBlenny/pages/pumpkins-vs-skeletons-game-for-circuitpython) game to use I2C gamepad input instead of USB gamepad input, the animation loop ended up running way faster. 

I'd like to find a way to speed up USB polling for synth and game projects so that more CPU time is available for `displayio` and `synthio` background tasks. Maybe this could be improved by using asyncio or something?

I'm posting this issue at tannewt's request to split out one of the points I mentioned in [USB Host Improvement Quest Umbrella Issue #10551](https://github.com/adafruit/circuitpython/issues/10551). The point of this issue is to let people know I'm planning to work on this myself and to ask for comments and feedback.

samblenny, 2025-08-08:
@tannewt For this one, I'd like to try profiling with GPIO pins and a logic analyzer. Do you have a favorite code snippet for quickly toggling pins from C?

tannewt, 2025-08-11:
> [@tannewt](https://github.com/tannewt) For this one, I'd like to try profiling with GPIO pins and a logic analyzer. Do you have a favorite code snippet for quickly toggling pins from C?

I use the pico-sdk gpio.h directly. Can't remember the API exactly. One CP trick is to ensure the pin number isn't reset. `never_reset` or adding a check in the reset function can do it.