adafruit/circuitpython #10556: TinyUSB submodule version is out of date

samblenny, 2025-08-08:
The current (10.0.0-beta.2) [TinyUSB @ 542e5b4](https://github.com/hathach/tinyusb/tree/542e5b4550a01d034b78308d77c408ed89427513) submodule is using a commit from 3 months ago.

There have been many TinyUSB commits with fixes and improvements since then.

dhalbert, 2025-08-08:
Thanks! A PR would be welcome. It would be good to do some testing on multiple platforms before merging the PR.

samblenny, 2025-08-10:
It's possible that the Pico-PIO-USB submodule (see [ports/raspberrypi/lib](https://github.com/adafruit/circuitpython/tree/main/ports/raspberrypi/lib)) should also be updated. It looks like its current submodule commit is 3 months old (v0.7.1 ?) and that the current version is v0.7.2.
Main change [appears to be](https://github.com/sekigon-gonnoc/Pico-PIO-USB/compare/032a469...main) this PR: https://github.com/sekigon-gonnoc/Pico-PIO-USB/pull/186

closed 2025-08-12 by samblenny