adafruit/circuitpython #10557: Fruit Jam HSTX/DVI output goes dark during web request

FoamyGuy, 2025-08-08:
This code illustrates the issue:
```
import time
import supervisor
import os

import adafruit_connection_manager
import board
import busio
from adafruit_esp32spi import adafruit_esp32spi
from displayio import Bitmap, TileGrid, Palette, Group
from digitalio import DigitalInOut
import adafruit_requests

BG_COLOR = 0x0000ff

display = supervisor.runtime.display

bg_bmp = Bitmap(display.width//4, display.height//4, 1)
bg_palette = Palette(1)
bg_palette[0] = BG_COLOR
bg_tg = TileGrid(bg_bmp, pixel_shader=bg_palette)

bg_group = Group(scale=4)
bg_group.append(bg_tg)
display.root_group = bg_group

# Get WiFi details, ensure these are setup in settings.toml
ssid = os.getenv("CIRCUITPY_WIFI_SSID")
password = os.getenv("CIRCUITPY_WIFI_PASSWORD")

# If you are using a board with pre-defined ESP32 Pins:
esp32_cs = DigitalInOut(board.ESP_CS)
esp32_ready = DigitalInOut(board.ESP_BUSY)
esp32_reset = DigitalInOut(board.ESP_RESET)

spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
radio = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)

print("Connecting to AP...")
while not radio.is_connected:
    try:
        radio.connect_AP(ssid, password)
    except RuntimeError as e:
        print("could not connect to AP, retrying: ", e)
        continue
print("Connected to", str(radio.ap_info.ssid, "utf-8"), "\tRSSI:", radio.ap_info.rssi)

pool = adafruit_connection_manager.get_radio_socketpool(radio)
ssl_context = adafruit_connection_manager.get_radio_ssl_context(radio)
requests = adafruit_requests.Session(pool, ssl_context)

FETCH_DELAY = 12

DATA_SOURCE = "https://cipher.foamyguy.com/static/size_tests/6363.json"

while True:
    #print("Fetching text from %s" % DATA_SOURCE)
    with requests.get(DATA_SOURCE) as response:
        response_text = response.text
        # print("-" * 40)
        # print("Text Response: ", response_text[:10])
        # print("-" * 40)

    time.sleep(FETCH_DELAY)
```

While the requests are being made the display output goes dark before going back to showing the blue from the background bitmap after the request completes. This one is in a loop so it will continually blink back and forth between blue and black on the display.

It seems like the SPI bus activity for the ESP32SPI web request might be blocking the display from working. But I am not sure how to validate that theory.

RetiredWizard, 2025-08-08:
I am seeing the same behavior on the Fruit Jam. I also see a slight variation of the issue with framebufferio displays, specifically HUB75 RGB matricies. When web requests are occurring the display flickers rapidly and random pixels light up giving a "static noise" like effect. 

Issue #10031 had similar symptoms but for for flash writes, not web requests....

RetiredWizard, 2025-08-08:
If the framebufferio issue is the same (and I'm thinking it's probably not now) then that's happening from an ESP32-S3 Matrixportal not an ESP coprocessor.

FoamyGuy, 2025-08-28:
I have not tried the smaller resolution specifically with the reproducer script but I can tomorrow. I have definitely seen the same issue appear with other projects that use the 320x240 resolution though. The say and spelling project I'm working on now uses that and shows this same behavior when downloading MP3s. 

The product overlay project that I've been working on with JP also uses 320x240 resolution and has the same blanking happen.

tannewt, 2025-08-29:
I think I've fixed this for the RP2350. The issue was that DMAing to or from PSRAM can stall out the DMA peripheral during a cache miss and throw off the timing of other DMA transfers. My fix makes SPI transfers use the CPU when one of the buffers is in PSRAM (or flash). This will potentially slow SPI transactions but ensures DMA transfers aren't stalled.