adafruit/circuitpython #10562: USB Host: Fast polling + write to CIRCUITPY crashes board

samblenny, 2025-08-09:
### CircuitPython version and board name

```python
Adafruit CircuitPython 10.0.0-beta.2 on 2025-07-30; Adafruit Fruit Jam with rp2350b
```

### Code/REPL

```python
import displayio
import gc
import time
import usb
from usb.core import USBError, USBTimeoutError

def test_descriptor_parsing_read_gamepad():
    # Check USB device descriptors, then read from gamepad if found.
    # This one is designed to trigger bugs where an in progress USB transaction
    # fights with other CircuitPython tasks. For me, this reliably triggers a
    # board reset with 10.0.0-beta.2, Fruit Jam rev D, and an 8BitDo SN30 Pro
    # wired gamepad (XInput style) or the Adafruit generic SNES style gamepad.
    # To trigger bug, attempt to write a file to CIRCUITPY once you see the
    # "Reading gamepad input..." message.
    gamepad_vid_pid_map = {
        # ( vid,    pid): (max_packet, sleep_before_read, description)
        (0x081f, 0xe401): (8, True, "Adafruit generic SNES style (HID)"),
        (0x045e, 0x028e): (32, False, "Xbox 360 compatible (XInput)"),
    }
    print("Finding USB devices...")
    try:
        for device in usb.core.find(find_all=True):
            # 1. Print descriptor properties
            vid = device.idVendor
            pid = device.idProduct
            print(f"idVendor      {vid:04x}")
            print(f"idProduct     {pid:04x}")
            print(f"serial_number {device.serial_number}")
            print(f"product       {device.product}")
            print(f"manufacturer  {device.manufacturer}")
            print(f"speed         {device.speed}")
            # 2. Configure interface for known gamepads (skip other devices)
            try:
                if not ((vid, pid) in gamepad_vid_pid_map):
                    print(" skipping this one (not a known gamepad)")
                    continue
                max_packet = gamepad_vid_pid_map[(vid,pid)][0]
                sleep_before_read = gamepad_vid_pid_map[(vid,pid)][1]
                buf = bytearray(max_packet)
                interface = 0
                if device.is_kernel_driver_active(interface):
                    device.detach_kernel_driver(interface)
                device.set_configuration(interface)
            except USBError as e:
                print(f"conf USBError: .errno={e.errno} '{str(e)}'")
            except USBTimeoutError as e:
                print(f"conf USBError: .errno={e.errno} '{str(e)}'")
            # 3. Rapidly poll gamepad for input. Goal of this is to trigger
            #    bugs where an in progress USB transaction fights with other
            #    CircuitPython tasks.
            print("Reading gamepad input...")
            print("Writing a file to CIRCUITPY now will probably reset board")
            while True:
                try:
                    if sleep_before_read:
                        # Adafruit generic SNES gamepad (low speed device) gets
                        # mad and raises USBError if you poll it too quickly
                        time.sleep(0.003)
                    n = device.read(0x81, buf, timeout=10)
                except USBError as e:
                    print(f"read USBError: .errno={e.errno} '{str(e)}'")
                except USBTimeoutError as e:
                    print(f"read USBError: .errno={e.errno} '{str(e)}'")
    except USBError as e:
        print(f"find USBError: .errno={e.errno} '{str(e)}'")
    except USBTimeoutError as e:
        print(f"find USBError: .errno={e.errno} '{str(e)}'")

displayio.release_displays()
gc.collect()

while True:
    # pause with a prompt to avoid rapidly scrolling serial console output
    input("press Enter to begin running test: ")
    print()
    test_descriptor_parsing_read_gamepad()
    print()
```

### Behavior

For me, the code above reliably triggers a bug where the board locks up when using an 8BitDo SN30 Pro wired gamepad (XInput style) or the Adafruit generic SNES style gamepad (HID style).

I've seen no indications that the bug depends on the model of gamepad is being used, or even that it's specific to gamepads. I mention those two because they are easy to obtain. My impression is that the bug is related to the frequency of calls to `usb.core.Device.read()`.

To trigger bug:
1. Run the code and connect the gamepad
2. Connect to the serial console and press Enter at the prompt to begin finding gamepad devices
3. Once you see the "Reading gamepad input..." message, attempt to write a file to CIRCUIPY

When I do this, the `copy code.py /Volumes/CIRCUIPY` shell command I'm using will get stuck (does not return to shell prompt). After a while, macOS gives me a "Disk Not Ejected Properly" notification and the serial console device disappears. The board LEDs do not indicate any error conditions. It seems like a pretty low level crash. To get things working again, I have to power cycle the board or press the reset button.

### Additional information

It's possible this might be related to https://github.com/adafruit/circuitpython/issues/10555 . That could be the case if the crash described here happens because input polling is starving other essential tasks for CPU time. (e.g. interrupt priority, blocking IO, or some other scheduling thing)

samblenny, 2025-09-04:
In shared-module/usb/core/Device.c, the functions that do endpoint and control transfers wait for their callbacks using a timeout loop that calls `RUN_BACKGROUND_TASKS;`. That in turn calls `tud_task()` and `tuh_task()` along with other CircuitPython background stuff. So, I suspect that whatever is going wrong is somewhere deeper in the TinyUSB implementation.

samblenny, 2025-09-09:
Also possible that calls to `device.read()` are generating so many `USBTimeoutError` exceptions that a significant fraction of CPU time is getting spent on the heap allocation and argument parsing involved in creating an exceptions. That could be tested by modifying `read()` to return 0 (no bytes transferred) instead of raising an exception.

samblenny, 2025-09-09:
Tried the reproducer with the current [absolute newest build (1a6c2b9)](https://adafruit-circuit-python.s3.amazonaws.com/bin/adafruit_fruit_jam/en_US/adafruit-circuitpython-adafruit_fruit_jam-en_US-20250909-main-PR10624-1a6c2b9.uf2), and it still acts the same way. (PR #10616 does not change this bug)

samblenny, 2025-09-09:
I tried adding `RUN_BACKGROUND_TASKS;` to every public method of `usb.core.Device`, including `read()`, but it made no difference. Still crashes the same way.

dhalbert, 2025-09-09:
I think it would be worth discussing this in https://github.com/hathach/tinyusb as an issue or in discussions to raise this for @hathach's attention. My understanding is that normally one call per tick should be fine, for this speed of USB. He may want to investigate further.

samblenny, 2025-09-10:
I'm gonna make a draft PR with the changes I've been trying today. But, @dhalbert, I think you're right that this needs a deeper look. With further testing, my fix just makes it so that instead of crashing every time, it will only crash some of the time.

samblenny, 2025-09-10:
I'm running low on ideas for how to proceed with this. If somebody else wants to take a look, go for it.

tannewt, 2025-09-10:
It looks like the background task is queued every host frame that needs it:
https://github.com/adafruit/circuitpython/blob/1a6c2b96db62dbe7e1dbec2b57d67f5778e20a69/ports/raspberrypi/common-hal/usb_host/Port.c#L64-L73

It also gets queued after USB interrupts:
https://github.com/adafruit/circuitpython/blob/1a6c2b96db62dbe7e1dbec2b57d67f5778e20a69/supervisor/shared/usb/usb.c#L219-L235

samblenny, 2025-09-10:
I remember seeing those and wondering if they might be related. How does one go about disabling PSRAM? Is there a config option for it?

samblenny, 2025-09-10:
That seems to totally fix the problem? I just tried a test build with this change:
```diff
diff --git a/ports/raspberrypi/boards/adafruit_fruit_jam/mpconfigboard.h b/ports/raspberrypi/boards/adafruit_fruit_jam/mpconfigboard.h
index 08c3271552..207babe13e 100644
--- a/ports/raspberrypi/boards/adafruit_fruit_jam/mpconfigboard.h
+++ b/ports/raspberrypi/boards/adafruit_fruit_jam/mpconfigboard.h
@@ -40,7 +40,7 @@
 #define DEFAULT_SD_CARD_DETECT (&pin_GPIO33)
 #define DEFAULT_SD_CARD_INSERTED true
 
-#define CIRCUITPY_PSRAM_CHIP_SELECT (&pin_GPIO47)
+// #define CIRCUITPY_PSRAM_CHIP_SELECT (&pin_GPIO47)
 
 // #define CIRCUITPY_CONSOLE_UART_TX (&pin_GPIO44)
 // #define CIRCUITPY_CONSOLE_UART_RX (&pin_GPIO45)
```

When I ran the reproducer code with an SN30 Pro gamepad, I was able to read files off CIRCUITPY without any trouble. Doing a `touch CIRCUITPY/boot.p` also triggered a normal auto-reload (rather than a crash).

dhalbert, 2025-09-10:
Great! The next step would be to see if there are PSRAM buffers being passed to TinyUSB or something like that, and fix as in #10598 or #10600.

samblenny, 2025-09-10:
Hmm... the `tuh_msc_*` stuff may be irrelevant. It doesn't seem like CircuitPython uses that section of the TinyUSB API.

I looked at the fixes in those two PR's but I've no clue how to go about applying similar changes to DMA that might be happening in the TinyUSB implementation. That stuff is a bit over my head. I'll step aside an leave this to other folks. I can test potential fixes though.

samblenny, 2025-09-15:
Just tested the CI build on Fruit Jam rev D. Works great.

closed 2025-09-22 by dhalbert