adafruit/circuitpython #10569: BusDisplay _refresh_area does not respect busio locks

JamesWhitlock, 2025-08-13:
If the RTOS is premptive, BusDisplay has a race condition that may caused faults on shared buses. displayio_display_bus_begin_transaction may return false indicating that the lock could not be acquired. As _refresh_area does not respect the return value of displayio_display_bus_begin_transaction, _send_pixels will try to write to the bus without a lock. A guard is present in the code, but it is implemented in a way that is not thread safe.

https://github.com/adafruit/circuitpython/blob/117b5d1ed9bab67f91f852777bf4cec8ee8b53e4/shared-module/busdisplay/BusDisplay.c#L290-L297

This may be responsible for issue #10536.

JamesWhitlock, 2025-08-13:
Further incorrect handling of displayio_display_bus_begin_transaction:
https://github.com/adafruit/circuitpython/blob/117b5d1ed9bab67f91f852777bf4cec8ee8b53e4/shared-module/epaperdisplay/EPaperDisplay.c#L165-L168

https://github.com/adafruit/circuitpython/blob/117b5d1ed9bab67f91f852777bf4cec8ee8b53e4/shared-module/displayio/bus_core.c#L131-L141

bablokb, 2025-08-16:
> I've spent a couple of hours trying to get a build environment working, but haven't succeeded

Have you tried using the build-system in the cloud? See https://github.com/adafruit/circuitpython/blob/main/.devcontainer/Readme.md for instructions.