adafruit/circuitpython #10590: M5Stack Cardputer bootloop with CircuitPython 10.0.0-beta.2

dglaude, 2025-08-27:
### CircuitPython version and board name

```python
Adafruit CircuitPython 10.0.0-beta.2 on 2025-07-31; M5Stack Cardputer with ESP32S3
Board ID:m5stack_cardputer
```

### Code/REPL

```python
No code
```

### Behavior

With ```Adafruit CircuitPython 9.2.0 on 2024-10-28; M5Stack Cardputer with ESP32S3``` the cardputer v1.0 (S3) and v1.1 (S3A) work fine.

But when upgraded to 10.0.0-beta2, with Open Installer, the board bootloop with the screen blinking with just the boot message, never appearing as a USB device (no CIRCUITPY drive).

### Description

[image]

### Additional information

This has been reported by todbot on 22/07/2025 08:32 with 10.0.0-beta0
And confirmed by me (David Glaude) 02/08/2025 14:49 on discord.

Depending scenario the bootloop display (blinking):
* Just the circuitpython logo
* Random pixels color
* The boot screen (just after flashing)

dhalbert, 2025-08-28:
10.0.0-alpha.8 is the first alpha that boot-loops.
10.0.0-alpha.7 comes up and show CIRCUITPY, but keyboard and serial input do not work.
10.0.0-alpha.2 is the last alpha that works as expected.

I will do some bisecting.

FYI @todbot

dhalbert, 2025-08-28:
A bisect was infeasible because between alpha.7 and alpha.8 was a merge from MicroPython with hundreds of unbisectable commits. However, I found that commenting out this in `mpconfigboard.mk`:
```makefile
SRC_C += boards/$(BOARD)/cardputer_keyboard.c
```
got rid of the boot-loop and allowed input to work again (at the expense of the on-board keyboard, of course). So I'll take a look at that code.

dhalbert, 2025-08-28:
The problem is that `common_hal_keypad_demux_demuxkeymatrix_construct()` allocates heap objects, but `cardputer_keyboard.c` wants to allocate an object that lives longer than a VM. So the constructor needs to be aware of this, probably in the style of `common_hal_lvfontio_ondiskfont_construct()`, which as a `use_gc_allocator` argument. I may want to generalize the allocation/deallocation routines used by lvfontio, since this is pretty much the same thing.

PR #10607 (2025-09-04): Fix cardputer boot loop, and do some cleanup
- Fixes #10590

Detailed changes:
- For `cardputer_keyboard.c`, allocate `DemuxKeyMatrix` it creates and the objects it holds (all the way down) on the port heap, because it lives past VM's.
- To do the above, add support for non-VM heap for `DemuxKeyMatrix` and the `keypad` allocations it needs by passing an arg `use_gc_allocator`, as is done in `lvfontio`.
- Add `mp_obj_port_malloc()`, `mp_obj_port_malloc_var()`, `mp_obj_new_port_tuple()`, all of which allocate on the port heap, not the VM heap.
- Add `port_malloc_zero()`, because `port_malloc()`  [... cut]