adafruit/circuitpython #10621: busio UART hardware flow control blocks control-C on raspberrypi

anecdata, 2025-09-09:
### CircuitPython version and board name

```python
Adafruit CircuitPython 10.0.0-beta.3 on 2025-08-29; Pimoroni Pico Plus 2 W with rp2350b
vs.
Adafruit CircuitPython 10.0.0-beta.3 on 2025-08-29; Adafruit Feather M4 Express with samd51j19
```

### Code/REPL

```python
import time ; time.sleep(3)  # wait for serial after reset
import sys
import board
import busio

def write(size, ch):
    print(f'writing {size} bytes...', end=" ")
    written = u1.write(f'{size * ch}')
    time.sleep(1)
    print(f'{written=} {} {u2.in_waiting=}')

if "Pico" in sys.implementation._machine:
    u1 = busio.UART(tx=board.GP0, rx=board.GP1, cts=board.GP2, rts=board.GP3)
    u2 = busio.UART(tx=board.GP4, rx=board.GP5, cts=board.GP6, rts=board.GP7)
elif "M4" in sys.implementation._machine:
    u1 = busio.UART(tx=board.A4, rx=board.A5, cts=board.D9, rts=board.D6)
    u2 = busio.UART(tx=board.A2, rx=board.A3, cts=board.D11, rts=board.D10)

u2.reset_input_buffer()
print(f'\nbusio initial {u2.in_waiting=}')

size = 64  # presumably fills busio default receiver_buffer_size=64 bytes
write(size, "A")

size = 32  # presumably fills raspberrypi hw uart rx buffer 32 bytes
write(size, "B")

if "Pico" in sys.implementation._machine:
    size = 6  # 5 somehow works; 6 will hang write and serial console
elif "M4" in sys.implementation._machine:
    size = 16  # can write as much as memory allows (e.g., 32768)
write(size, "C")

r = u2.read(256)
print(f'read {r} {len(r)} bytes')
```

### Behavior

Pico output (writing 102 bytes):
```
busio initial u2.in_waiting=0
writing 64 bytes... written=64 () u2.in_waiting=64
writing 32 bytes... written=32 () u2.in_waiting=64
writing 6 bytes... 
```
• The write hangs, but also the serial console hangs. CIRCUITPY is still accessible. Reset is required to regain use of the serial console.

or

Pico output (writing 101 bytes):
```
busio initial u2.in_waiting=0
writing 64 bytes... written=64 () u2.in_waiting=64
writing 32 bytes... written=32 () u2.in_waiting=64
writing 5 bytes... written=5 () u2.in_waiting=64
read b'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABBBBBBBBBBBBBBBBBBBBBBBBBBBBBBBBCCCCC' 101 bytes
```
• Flow control limit not reached: bytes written == bytes read.
• Note that `in_waiting` only counts the `receiver_buffer_size`, not the pico hardware buffer or the magic extra 5 bytes.

_...compare to..._

M4 output:
```
busio initial u2.in_waiting=0
writing 64 bytes... written=64 () u2.in_waiting=64
writing 32 bytes... written=32 () u2.in_waiting=64
writing 16 bytes... written=16 () u2.in_waiting=64
read b'AAAAAAAAAAAAAAAABBBBBBBBBBBBBBBBBBBBBBBBBBBBBBBBCCCCCCCCCCCCCCCC' 64 bytes
```
• Hardware flow control doesn't seem to work. Any number of bytes can be written and the buffer will contain the last 64 bytes sent.

tannewt, 2025-09-09:
Is your goal of this issue to fix ctrl-c? I suspect it is due to the underlying SDK UART calls blocking rather than CP's loop to wait. It is possible that our waiting loop isn't checking to see if it is interrupted.

anecdata, 2025-09-09:
I see checks like that in the `read`s, but not in the `write`s on any of the three ports

`atmel-samd` specifically has a code comment:
> non-blocking and will return immediately after setting up the write

but `atmel-samd` is the odd one that allows you to keep writing after the buffer is full with hardware flow control pins defined (new bytes shifted in, old bytes shifted out)

samblenny, 2025-09-09:
Is there some other code in a different place that handles transferring buffered write data to the UART hardware?

samblenny, 2025-09-09:
If it is an SDK level call, those would probably not be checking `mp_hal_is_interrupted()`. Maybe that's the problem?

closed 2026-10-19 by dhalbert