adafruit/circuitpython #10632: SAMD51: mounting SD card causes crash

dhalbert, 2025-09-14:
Between 10.0.0-alpha.4 and alpha.5, this `code.py` started crashing on SAMD51, specifically PyPortal:

```
import board
import sdcardio
import storage
import time

spi = board.SPI()
cs = board.SD_CS

print("mounting sd card...")
sdcard = sdcardio.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")
print("sd card mounted")

while True:
    print("sleep 1")
    time.sleep(1)
```
It prints `sleep 1` once and then hangs.

I'm testing on other ports as well. On an RP2040 Metro it does not crash, but often I get "no SD card", and it requires a hard power-cycle. Also on RP2040, a drive for the SD card appears on Linux, but is not mounted.

#10351 is a possibly relevant PR that was included in alpha.5. I am thinking this all has to do with auto-USB-presentation of secondary mounts.

dhalbert, 2025-09-21:
### Summary of testing

#### PyPortal (SAMD51): mount SD manually card in `code.py`, and then print a count once per second
- After only one or two seconds, the program stalls. CIRCUITPY never appears. Eventually the program starts up again, as visible on the display, and there are one or more reconnects/disconnects on USB serial.
  - A USB trace shows that there is occasional USB MSC activity (SCSI status commands) for tens of seconds. Eventually there is some fatal error in Linux USB subsystem and attempts to see PORTALBOOT on various USB ports fails. Linux has to be rebooted.
- I enabled USB debug logging but didn't see anything unusual going on -- no reports of assertion failures, etc.
- I added a compile-time option to turn on/off presenting the SD card as a LUN. When that option is turned off, the PyPortal works fine.
#### Metro ESP32-S3:  mount SD card manually in `code.py`, and then print a count once per second
- CIRCUITPY and the SD card appear as drives. On Linux, there are several 70-second intervals at the end of which there is an auto-reload, apparently due to a filesystem write. Eventually this seems to stop. These intervals don't happen on Windows 11 or macOS.
#### RP2350 (Metro and Fruit Jam): automount of SD card; `code.py` only prints a seconds counter
- CIRCUITPY and SD card appear. No 70-second interruptions.

My inclination is to turn off SD USB for SAMD51 for now, for 10.0.0, and leave an open issue for it to be fixed later.
Whether it should be turned off for Espressif is another question: the auto-reloads are annoying but could be turned off.

tannewt, 2025-09-22:
It makes no sense to me that there would be 70 second autoreloads on one platform only since it feels like host behavior. 

Disabling on SAMD is fine with me since that could be a lower level USB bug.

PR #10645 (2025-09-26): Sd card usb presentation fixes
- Fixes #10632.
- Does _not_ deal with #10510, which touches some of the same code, but I'll put that in a separate PR after some further consultation.
---
#### SD card USB presentation
- Add new `.h`-only compile option `CIRCUITPY_SDCARD_USB`, which will enable/disable presenting SD card as a USB MSC LUN.
- Enable on all ports except `atmel-samd` due to some bug that causes USB disconnection. I tested most of the other ports, and they don't have this problem. I'll open a separate issue about this for fixing post 10.0.0.
#### Prevent unneeded auto-reload
- Do not [... cut]