adafruit/circuitpython #10639: Code randomly does not invoke Magtag..exit_and_deep_sleep()

gglockner, 2025-09-22:
### CircuitPython version and board name

```python
CircuitPython 10.0.0-beta.3, MagTag 2025 edition
```

### Code/REPL

```python
from adafruit_magtag.magtag import MagTag
from os import getenv
import gc

url = getenv("PWD_URL")
print("Getting data from URL "+url)
magtag = MagTag(
  url=url,
  json_path=[[], [], ["ssid"], ["passphrase"], ['timestamp'], ['qrcode']]
)

def update(magtag):
  try:
    result = magtag.fetch(auto_refresh=False)
    print("Retrieved data: "+str(result))
    magtag.graphics.qrcode(result[-1], qr_size=3, x=192, y=0)
    magtag.refresh()
    sleepTime = getenv("REFRESH_TIME")
  except Exception as e:
    print("Error-callback:\n", str(e))
    sleepTime = 60
  
  gc.collect()
  return sleepTime

title_font = getenv("TITLE_FONT")
header_font = getenv("HEADER_FONT")
body_font = getenv("BODY_FONT")
small_body_font = getenv("SMALL_BODY_FONT")

magtag.add_text(
    text_font=title_font,
    text_transform=lambda x: "Guest Wi-Fi",
)

magtag.add_text(
    text_font=header_font,
    text_position=(0,20),
    text_transform=lambda x: "Scan to connect",
)

magtag.add_text(
    text_font=body_font,
    text_position=(0,100),
    text_transform=lambda x: "Net:  "+x,
)

magtag.add_text(
    text_font=body_font,
    text_position=(0,115),
    text_transform=lambda x: "Pass: "+x,
)

magtag.add_text(
    text_font=small_body_font,
    text_position=(0,70),
    text_transform=lambda x: "Update: "+x,
)

sleepTime = update(magtag)
print("Entering deep sleep for "+str(sleepTime)+" sec")
magtag.exit_and_deep_sleep(sleepTime)
```

### Behavior

Normally, when the code finishes, the console writes:

```
Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
Pretending to deep sleep until alarm, CTRL-C or file write.
Woken up by alarm.
```

However, after several hours, you see:

```
Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

and the board is frozen, without invoking deep sleep.

gglockner, 2025-09-22:
@dhalbert : I can reproduce this when the Magtag is disconnected from the computer. I just posted the console log for illustration.

I am testing a much simpler version below. The board is currently disconnected from the computer. Will post an update in a few hours when I have tested it more.

```
from adafruit_magtag.magtag import MagTag
from os import getenv

url = getenv("PWD_URL")
print("Getting data from URL "+url)
magtag = MagTag(
  url=url,
  json_path=[['timestamp']]
)

title_font = getenv("TITLE_FONT")

magtag.add_text(
    text_font=title_font,
    text_position=(20,20),
)

result = magtag.fetch()

sleepTime = 60

print("Entering deep sleep for "+str(sleepTime)+" sec")
magtag.exit_and_deep_sleep(sleepTime)
```

dhalbert, 2025-09-22:
If the `magtag.fetch()` fails, I think you will get an (uncaught) exception. So you could catch all exceptions and try again or just go to sleep.

gglockner, 2025-09-22:
I can confirm that it fails with the simpler code above, even with the Magtag disconnected from a computer.

dhalbert, 2025-09-22:
Could you try 9.2.9 as well? Thanks. It would be useful to know if this is a regression.

gglockner, 2025-09-22:
> OK, so not even a fetch is involved. When you say "a few hours", is that time actually consistent?

I don't know. It takes too long to babysit.

> Could you try 9.2.9 as well? Thanks. It would be useful to know if this is a regression.

Unfortunately no, according to [the instructions for installing CircuitPython](https://learn.adafruit.com/adafruit-magtag/circuitpython):

> WARNING: The updated Adafruit MagTag 2025 Edition will not work with CircuitPython 9.2.x or earlier. Make sure you install 10.x.x or later!

So I cannot run CircuitPython 9.2.9 for my MagTag 2025 Edition.

dhalbert, 2025-09-22:
Thanks anyway, I have an older MagTag and can test.

gglockner, 2025-09-22:
> When you say "a few hours", is that time actually consistent?

I modified the test so that it is deterministic. See below. I confirmed the number of iterations is not consistent: I got a failure of 61 and 16 trials, on the same MagTag. If you want to start with a fresh count each time, you need to run the two commented lines via REPL.

```
from adafruit_magtag.magtag import MagTag
import terminalio
import microcontroller

## INIT NVM via REPL
# i = 0
# microcontroller.nvm[0:8] = i.to_bytes(8)

i = 1+int.from_bytes(microcontroller.nvm[0:8])

magtag = MagTag()

magtag.add_text(
    text_font=terminalio.FONT,
    text_position=(20,20),
    text="Trial "+str(i)
)

microcontroller.nvm[0:8] = i.to_bytes(8)

magtag.exit_and_deep_sleep(2)
```

gglockner, 2025-09-25:
Good news, I found the root cause: if the EPaperDisplay is updating during the call to `MagTag.exit_and_deep_sleep()`, it may fail to restart the code. I just created [pull request 96](https://github.com/adafruit/Adafruit_CircuitPython_MagTag/pull/96) for the MagTag library, which solves the issue by going into light sleep while waiting for the display to finish updating.

closed 2025-09-25 by gglockner

dhalbert, 2025-09-25:
Thanks for the diagnosis! I think this is still a bug. You should not have to depend on the library to fix the problem, since you may not even be using the library. `exit_and_deep_sleep()` should wait for the display update or cancel it.

gglockner, 2025-09-27:
@bablokb : After some thought, I agree that it would be cleaner to make a blocking option on `MagTag.refresh()` instead. Specifically, if `MagTag.refresh(blocking=True)` then the function waits to return until the display has finished updating. That wouldn't change the behavior of exit and deep sleep. I just added another commit to my pull request for [MagTag](https://github.com/adafruit/Adafruit_CircuitPython_MagTag/pull/96).

dhalbert, 2025-10-03:
@tannewt What do you think is the right thing to do here?
https://github.com/adafruit/Adafruit_CircuitPython_MagTag/pull/96 fixes the library, but not the general problem. I haven't yet looked to see if it would be possible to wait for the refresh before shutting down. I'm not sure we need a blocking option in the call, but we do need to block before shutdown.